ouster-lidar/ouster-sdk #102: cannot visualize data on rviz

aaravrav142, 2019-08-21:
Hi, 

I am trying to visualize the data on rviz using the bag files provided. However I can see nothing on the rviz. How do I solve this problem.

Steps I am following:
1. Compiled the ouster_example code into my catkin_ws
2. Compiled the viz inside the ouster_viz by mkdir build && cmake .. && make 
3. Downloaded bag files from the link provided in the page 
4. run the ouster ros node $ roslaunch ouster_ros os1.launch replay:=true
5. open rviz with the correct config file inside ouster_ros 
6. play the bag file with --clock parameter.

I can see the topics when I do rostopic list but nothing is being published when I echo points 

Thanks

dsb6063, 2019-08-21:
If there is no output using echo, then you are not publishing into a topic. Did you install the other parts of the ouster_ros packages? If you do not have a sensor, then you may not need ouster_client. Try to install the Ouster_Ros package, since you installed the ouster_rviz. 

*	ouster_client/ <https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/README.md>  contains an example C++ client for the OS-1 sensor
*	ouster_viz/ <https://github.com/ouster-lidar/ouster_example/blob/master/ouster_viz/README.md>  contains a visualizer for the OS-1 sensor
*	ouster_ros/ <https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/README.md>  contains example ROS nodes for publishing point cloud messages

Your command #4 invokes ouster_ros package, if you have not installed that portion, that could be the issue. I attached the above snippet from the readme. Also, remember to source the package setup. Search the forum for this item, you should find several examples. 

If you have installed all of the above, make sure the bagfile is executable by chmod +x. The Ouster Studio will help with live sensor to verify if your connections are working. 

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

aaravrav142, 2019-08-26:
Hi @dsb6063,

Thank you for the reply. I do not have the real lidar unit with me, just testing with the bag files. 

I have all the packages mentioned by you and build properly inside my catkin_ws.

The problem is with the version of the ouster package. Somewhere down the line I think the packages were updated with new version and I cannot play the data on rviz or the ouster viz. 
I tested the same bag files with the previous version of the ouster driver for ros (https://github.com/csm-forklift/ouster_example) and I can see the data with this particular driver. 

Can you provide a new bag file to test with the laster commit. 

Thanks.

mrgransky, 2020-01-07:
Instruction how to visualize ouster lidar with recorded data:

Terminal 1:
```
$ cd ~/catkin_ws
$ source devel/setup.bash
$ roslaunch ouster_ros os1.launch replay:=true metadata:=path/2/sample-sensor.json image:=true
```

Terminal 2:
`$ rosbag play --clock -l path/2/file.bag`

closed 2020-03-14 by dgarbuzov