ouster-lidar/ouster-sdk #103: ouster_ros with .bag file "hostname"?

mrgransky, 2019-08-21:
Hi,
I am trying to work with ouster_ros from already recorded sample sensor bag file provided [here](http://data.ouster.io/sample-data-1.12/index.html), but in the following command, it asks for `hostname` which I do not know what to provide.

`roslaunch ouster_ros os1.launch replay:=true os1_hostname:=<os1_hostname>`

Cheers,

mohit-khosti, 2019-08-21:
hostname is the IP address of the Ouster LiDAR sensor

mrgransky, 2019-08-21:
so what's that IP address ?
I only get .bag file! no LIDAR sensor

mrgransky, 2019-09-10:
@zuzi-m  Thank you very much. 
It helped me a lot.

zuzi-m, 2019-09-11:
@mrgransky if you don't have any more questions or problems regarding this issue, you can close to mark it as resolved :)

closed 2019-09-12 by mrgransky