ouster-lidar/ouster-sdk #104: Quality of Intensity/Range/Noise images

dcazzato, 2019-08-26:
Hello,

I am able to receive and visualize the intensity, noise and range images while operating with the ROS node. 

Obtained output has many horizontal lines around any vertical shape of the images, with much worst quality than in the data published in your videos. As an example, I attach some image (in order intensity, noise and range image). For example, the black border where I added an arrow has a very tangible sawtoothed shape.

Is there any way to improve this output, is it a sensor issue? Please note that I already tried with different lidar_mode (these images are taken while working at 1024x20), scan_dur_ns, as well as with Ouster Studio.

Thank you,

[image]
[image]
[image]
[image]

dcazzato, 2019-08-28:
Hi, first of all, thank you for your reply. I acquired the same 3 images outdoor and results are much better in terms of intensity and depth. Anyway, the noise image has still an appearance like in the image attached. I am wondering if this output is standard.

[image]

dgarbuzov, 2020-03-14:
That could be normal depending on the environment. We'll be releasing some sample code for post-processing the ambient and intensity images to address some of these issues soon.

closed 2020-03-14 by dgarbuzov

TongxingJin, 2024-06-02:
> That could be normal depending on the environment. We'll be releasing some sample code for post-processing the ambient and intensity images to address some of these issues soon.

Hi, have you ever released something to show how to post-process the ambient and intensity images up to now?

Samahu, 2024-06-03:
@TongxingJin I think dmitrig was referring to the [image_processing module](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/image_processing.h) please check it it out.

TongxingJin, 2024-06-06:
> @TongxingJin I think dmitrig was referring to the [image_processing module](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/image_processing.h) please check it it out.

Hi Samahu, thanks for your reply, it helped me a lot. I think the laser emitters have different magnitudes of power and the corresponding receivers have different levels of sensitivity, that's the main reason for the artifact line in different rows of ambient and intensity images, right?

patripfr, 2024-11-08:
@TongxingJin I have also developed a tool to remove these line artifacts from the ouster images, you can check it out here:
https://colab.research.google.com/github/ethz-asl/COIN-LIO/blob/main/scripts/artifact_removal.ipynb
More info about it can be found in the related publication: https://arxiv.org/abs/2310.01235