ouster-lidar/ouster-sdk #107: .rosbag to .pcap

pcarrascof, 2019-09-05:
I'm using Ouster Studio so I want to convert my own rosbag to .pcap files to examinate it in that program.
Can it be posible?

wangzhensuo, 2020-09-09:
I have the same question. only find method to get pcd file from bag,and didnot find how to get pcap file from bag.
rosrun pcl_ros bag_to_pcd my.bag /points_raw pcd/

sam-rke, 2021-01-15:
Hi @pcarrascof  , @wangzhensuo , @dgarbuzov  !!
do you guys find how to convert .bag file to .pcap file.
I can fine how to get a .bag file but I am stuck at .pcap generation, any help would be grateful.
Thanks

wangzhensuo, 2021-01-16:
@sam-rke   hi, I also meet the same problem while importing pcap file to matlab.
search for a long time on internet,   did not find related method.it seems no way. 
one .bag file can  generate several pcd files, but one pcap file contains several pointcloud iamg.
maybe need some formula  to transfer.
I don't have hardware sensor,it seems pcap file could be got from sensor directly.

sam-rke, 2021-01-18:
HI @wangzhensuo 
actually pcap can be generated from Ouster Studio (as per Ouster Guidline)
I'll be looking into it as I'll be getting Ouster Lidar soon.
Hope for the best.
Thankyou

kairenw, 2022-02-03:
Hi @sam-rke, just so you know: the Python SDK records .pcap files, as shown in [this example](https://static.ouster.dev/sdk-docs/examples.html#recording-sensor-data). If you have other questions, 

Providing bag to pcap functionality is pretty low on our list of features, so I'm marking this as won't do. Given that most of the requests in this thread desired bag to pcap so that people could use Ouster Studio to examine their data, I would recommend checking out Python SDK if you don't need bags. If you need to record bag data, but also need it in pcap format, you can always use `tcpdump` to record pcaps while recording the bags. Hope that helps!

closed 2022-02-03 by kairenw