ouster-lidar/ouster-sdk #114: Setting static IP address

petrlmat, 2019-09-16:
Hi,
I am trying to set a static IP for the Ouster OS1 by issuing a PUT request using httpie according to the [software user guide](https://data.ouster.io/downloads/software-user-guide-v1.12.0.pdf).
```
http -v PUT http://10.5.5.99/api/v1/system/network/ipv4 override=10.5.5.99/24
```

However, the sensor responds with 405 error:
```
PUT /api/v1/system/network/ipv4 HTTP/1.1
Accept: application/json, */*
Accept-Encoding: gzip, deflate
Connection: keep-alive
Content-Length: 28
Content-Type: application/json
Host: 10.5.5.99
User-Agent: HTTPie/0.9.8

{
    "override": "10.5.5.99/24"
}

HTTP/1.1 405 Method Not Allowed
content-length: 46
content-type: application/json; charset=UTF-8

{
    "error": {
        "title": "405 Method Not Allowed"
    }
}
```

The error message is the same when I tried using POST request, and also when using curl instead of httpie. Can you please help me set a static IP to the sensor?

Thanks!

zuzi-m, 2019-09-17:
I think you are only missing the last part of the API route:
In your example, you are making a following request: `PUT system/network/ipv4`
Documentation says that the route is `PUT system/network/ipv4/override`

Also, then the data passed to this route should be just a string with the IP address, not a JSON object.

petrlmat, 2019-09-17:
If I understand it correctly, the request should then be in this form?
```
http -v PUT http://10.5.5.99/api/v1/system/network/ipv4/override "10.5.5.99/24"
```
Then it seems like it is not even a valid request.
```
usage: http [--json] [--form] [--pretty {all,colors,format,none}]
            [--style STYLE] [--print WHAT] [--headers] [--body] [--verbose]
            [--all] [--history-print WHAT] [--stream] [--output FILE]
            [--download] [--continue]
            [--session SESSION_NAME_OR_PATH | --session-read-only SESSION_NAME_OR_PATH]
            [--auth USER[:PASS]] [--auth-type {basic,digest}]
            [--proxy PROTOCOL:PROXY_URL] [--follow]
            [--max-redirects MAX_REDIRECTS] [--timeout SECONDS]
            [--check-status] [--verify VERIFY]
            [--ssl {ssl2.3,tls1,tls1.1,tls1.2}] [--cert CERT]
            [--cert-key CERT_KEY] [--ignore-stdin] [--help] [--version]
            [--traceback] [--default-scheme DEFAULT_SCHEME] [--debug]
            [METHOD] URL [REQUEST_ITEM [REQUEST_ITEM ...]]
http: error: argument REQUEST_ITEM: "10.5.5.99/24" is not a valid value

```

zuzi-m, 2019-09-17:
Sorry, I didn't notice your example uses httpie.

In this case, you have to write the value you want to set into the request body. The problem is that the value that this endpoint accepts is not a json object but just a string so you can't do it with httpie's handy parameters as you tried before (they require a key=value format which creates a json object)

What you have to do is write the values directly into the request body. That is done by writing it to standard input (explained [here](http://igorsobreira.com/2016/10/19/post-string-using-httpie.html) )

So try this
```
echo '"10.5.5.99/24"' | http -v PUT http://10.5.5.99/api/v1/system/network/ipv4/override
```

If that fails, maybe the api needs the correct content-type to be set, so try adding `--json` flag to the `http` command

petrlmat, 2019-09-18:
Thanks for the help! This indeed worked:
```
echo '"10.5.5.99/24"' | http -v PUT http://10.5.5.99/api/v1/system/network/ipv4/override
```

closed 2019-09-18 by petrlmat

heuristicus, 2021-11-11:
If you want to do this with `curl` it is

```
curl -X PUT -H "Content-Type: application/json" 10.5.5.99/api/v1/system/network/ipv4/override -d '"10.5.5.99/24"'
```

yuzhoul1991, 2022-03-03:
Does this static ip still in effect when the lidar reboots? If not, iss there a way to set a static IP premaritally?