ouster-lidar/ouster-sdk #119: Trabsforming Lidar coordinate to Sensor frame

Chenweng1, 2019-10-21:
Hi, Can anyone please share the process of transforming the lidar coordiante into the sensor coordinate so that we can get the real world coordinates? If I run the command on TCP like 
 "get_lidar_intrinsics". I get the output like this

{"lidar_to_sensor_transform": [-1, 0, 0, 0, 0, -1, 0, 0, 0, 36, 18, 0, 0, 0, 1]}.

It's a 4x4 matrix while the Lidar data coordinates is 3x1like the (x,y,and z)

Any help will be very appreciated in this matter.

Thanks

bromeara, 2019-11-22:
You need to use the [tf library](http://wiki.ros.org/tf) in ROS and you need to look that up on your own as it is a big piece of software. But the transforms are already published by the [os1_cloud_node](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/os1_cloud_node.cpp) so you would need to use tf to transform the data on the /os1_cloud_node/points topic from the /lidar_frame to the /sensor_frame.

dgarbuzov, 2020-12-07:
The meaning of the matrix is explained in the user manual: https://data.ouster.io/downloads/software-user-manual/software-user-manual-v2p0.pdf#e (section 3.1.6):

It's a row-major transformation matrix in homogenous coordinates. Since the latest update to the example code, this transformation is applied by default: it's passed as the "transform" parameter to `make_xyz_lut`:

https://github.com/ouster-lidar/ouster_example/blob/9352730127c3e2bc8ce1e778be4ae8b0a3b40f3c/ouster_client/src/lidar_scan.cpp#L13

closed 2020-12-07 by dgarbuzov

hichamhendy, 2022-02-02:
If I mount the LiDAR  with angle 180 or another angle , how can export that so that I can have os_sensor orientated parallel to the   ground?

I tried to change the meta.json However it got changed through the the package!