ouster-lidar/ouster-sdk #120: Bad data sensor os1 readings

pamarinp, 2019-10-31:
Hello, 

Currently we have the lidar mounted on a vehicle and we have experienced several cases when the data is corrupted in a random zone. We are using ouster_example launchfile with all the default parameters. The configuration is at 10Hz.

Here are 2 consecutive readings:

https://ibb.co/nQ9sMyX

https://ibb.co/0fNDN26

zuzi-m, 2019-11-04:
What exactly do you mean by corrupted data? Is there data missing in the intensity/noise/range images in visualization, or is only the range missing?

If only range is missing, it can be the case of [a known issue described here.](https://github.com/ouster-lidar/ouster_example/issues/96)

If everything is missing in sections of the lidar data, it might be because the lidar sends data over a UDP network protocol which is unreliable - parts of the data are not guaranteed to arrive in the same order they were sent in, and they are not even guaranteed to be delivered at all. So it might be a case of the data packets arriving in a different order, some packets missing, or some being dropped  on the way.

bromeara, 2019-11-21:
We have experienced similar issues and it seems to indeed be due to the UDP connection after analyzing our bag files, Are these issues that new UDP_TRANSMISSION alert could be used to detect when they're occurring or is the alert more for detecting UDP disconnects?

HappySamuel, 2020-02-11:
Hi all,

I also faced the same issue with my OS1-16, below are the images of a stream of video
[image]

[image]

[image]

![Uploading Screenshot from 2020-02-11 16-09-01.png…](
[image]

[image]

[image]

zuzi-m, 2020-02-11:
@HappySamuel it is always a good idea to upgrade the firmware and if your problem is really related to the known firmware issue then yes, it is fixed in version 1.13. There are no breaking changes between 1.12 and 1.13 so you can upgrade without having to make any other changes

HappySamuel, 2020-02-11:
Hi @zuzi-m 

I have upgraded the os1-16 firmware to 1.13, and the above issue somehow improve, i no longer lost so much points. However the issue still persists, i saw someone named it as propeller effect. And this propeller effect only appear when the lidar mode is 2048x10 or 1024x20, any mode lower than this operate normally. Attached is the videos i recorded, hope they can explain the situation.

1024x20
[image]

2048x10
[image]

1024x10
[image]

512x20
[image]

Please advice how shall i get rid of this propeller issue, ensuring there is no lidar packet loss?

Best,
Samuel

adityarath302, 2020-03-04:
The issue is with your Ethernet .I had the same issue and once i changed the Ethernet port to Gbps one it stopped giving that propeller effect.

HappySamuel, 2020-03-04:
@adityarath302 

I have been using cat7 Ethernet cable from ouster lidar box to the gigabyte network switch, as well as cat7 Ethernet cable from network switch to my PC. But the problem still persist.

Best, 
Samuel

adityarath302, 2020-03-04:
Check your NIC ,does it support Gbps Ethernet.

adityarath302, 2020-03-04:
The cat7 wires are fine,this is a data loss issue due to hardware not being able to capture the required points.Some hardware in between lacks Gbps functionality.Please try connecting the Lidar to the systems directly and see the results.

HappySamuel, 2020-03-04:
Hi @adityarath302 

I will try to connect it to my system directly and see whether does this issue persist or not. 
Btw, what's NIC?

Best,
Samuel

adityarath302, 2020-03-04:
Network Interface card

dgarbuzov, 2020-03-13:
Even with the correct hardware, your interface may fail to negotiate a gigabit link. You can check the actual link speed with a tool like `ethtool`.

closed 2020-03-13 by dgarbuzov

HappySamuel, 2020-03-16:
Hi @dgarbuzov 

If my interface fail to negotiate a gigabit link, is there any solution for this?? As i have 3 same configuration, 2 of them doesn't have this propeller issue. But this one has it.

Best,
Samuel

dgarbuzov, 2020-03-16:
Try swapping interface boxes, cables, and nics to isolate the problem. If you think the issue is with one of the interface boxes, contact ouster support for a replacement.