ouster-lidar/ouster-sdk #121: How to eliminate sim_time in msg stamp

zlg9folira, 2019-11-03:
Hi, 

When I fire up the OS-1 and `echo` topics like points or imu, it looks that the time stamp is `sim_time` and not a 10 digit posix/unix time (which I need). 

sec 4583 | nsec 855689180

I checked `/use_sim_time` parameter and noticed that it was not shown under `rosparam list`. So I configured it `rosparam set /use_sim_time false` and again fired up the node. I am still getting a very short time stamp which is `sim_time`. Any workaround to get rid of this stamp format?

zuzi-m, 2019-11-04:
This is because the `os1_cloud_node` sets the time stamp of the messages published in its topics to the time stamps that come in the lidar packet data from the sensor. The values there come from the sensor itself and not from ROS.

If you want to use the ROS time as time stamp of the topics, you will have to modify the code yourself or wait for Ouster or someone else to make the changes. If you implement the change yourself you can open a pull request in this repository with the changes so that other users could benefit from it too :)

zlg9folira, 2019-11-04:
In `os1_ros.cpp`, i changed the stamp setting line from `msg.header.stamp.fromNSec(timestamp.count());`  TO  `msg.header.stamp = ros::Time::now();` .

However, this didn't solve the issue. I also checked the launch file and other files, couldn't find any other thing related to time-stamp setting.

zlg9folira, 2019-11-04:
Yeah, thanks for the response. I have opened a pull request on this with the updates.

closed 2019-11-04 by zlg9folira

asherikov, 2019-11-05:
> sets the time stamp of the messages published in its topics to the time stamps that come in the lidar packet data from the sensor.

This is the correct way to do this. Using `ros::Time::now()` is wrong, you should configure time synchronization instead.