ouster-lidar/ouster-sdk #123: Range data offset

dipendras, 2019-11-06:
Hi,

I have tried reading the x,y,z coordinates with respect to lidar_frame of the pointcloud obtained from OS1 lidar. The data seems to have offset in specially y-coordinates of the points. I have tried subtracting 100mm in the raw range data to see if it can improve the point coordinates, it seems that the readings are better than before. Is there a way to correct/calibrate the depth measurements (range or x,y,z coordinates offset with respect to lidar_frame)?

bromeara, 2019-11-21:
You are looking for the [tf library](http://wiki.ros.org/tf), the frame you actually want to be working in is the sensor_frame which is specified in the OS1 Software Users Guide as being at the base of the lidar. The transform is published by the os1_cloud_node.cpp and you should be able to see what the actual values are by doing `$ rostopic echo /tf_static`. The tf library is complicated and outside of the scope of this library but you would use it to transform your lidar data from the lidar_frame to the sensor_frame.

dgarbuzov, 2020-03-11:
Please contact ouster support with any suspected device calibration issues

closed 2020-03-11 by dgarbuzov