ouster-lidar/ouster-sdk #126: Estimating Cartesian Coordinate from the Range Data

Chenweng1, 2019-11-13:
HI, Can anyone please tell me does the "rosrun pcl_ros point cloud to pcd" tool considers the beam angles (altitude, azimuth angles, and the range value) while creating the Cartesian coordinates of the point clouds ?

Thanks 
Weng

bromeara, 2019-11-21:
This is separate from the pcl_ros pointcloud_to_pcd node the data is actually published to the /os1_cloud_node/points topic in the cartesian coordinates. The code for which is found in [os1_util.h](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/os1_util.h) in the batch_to_iter function.

dgarbuzov, 2020-12-07:
The published pcl point cloud topics are already in cartesian coordinates and take beam angles into account

closed 2020-12-07 by dgarbuzov