ouster-lidar/ouster-sdk #127: Frame Serialization Issue

ethan20120, 2019-11-16:
Hi,
Does the ros maintain the serial of the LiDAR frame while creating the PCD file from the ros bag by running the "pointcloud to pcd" node ?

Thanks

bromeara, 2019-11-21:
I'm unsure what you mean by "serial of the LiDAR frame" but if you mean the frame ID then the answer is no although since the packets from one frame are batched into one pointcloud2 message the header/seq field of the pointcloud2 message is basically frame relative to when the ros node was started.

If you mean the raw data, which are UDP packets, then the answer is also no because pointcloud_to_pcd only subscribes to the /os1_cloud_node/points topic (because it publishes pointcloud2 messages), whereas the UDP messages are broadcasted on the /os1_node/lidar_packets topic with the /PacketMsg message type.

dgarbuzov, 2020-12-07:
If I understand the question correctly, no the frame id is currently not published in ROS messages, though I can see how that would be useful. If there's a simple way to do it while preserving compatibility with standard ROS/PCL messages, please submit a PR!

closed 2020-12-07 by dgarbuzov

bromeara, 2020-12-09:
I forked the example_library a while ago and it looks like there has been substantial changes and I have no need for the frame IDs in my messages. But the easy way would probably just be to pack it into the header/seq field since it is normally just an arbitrarily incremented int in ROS1 anyways.