ouster-lidar/ouster-sdk #128: What is the range of intensity and reflectivity?

maskjp, 2019-11-22:
Hi, I have a question about the range of intensity and reflectivity. I saw the code, which generates the intensity image, set all values larger than 255 to 255. Does that mean the range of intensity and reflectivity is from 0 to 255? But the intensity image seems to saturate a lot. And the value of reflectivity and intensity seems equal.

Besides, what does the noisy image mean? 

Thank you!

joelsa, 2019-11-25:
Intensity returned from the OS-1 is a 16-bit Integer, so ranging from 0 to 65535. In practice, I have found really bright retroreflectors to have values up to 6000, whereas most "normal" surfaces lie in between 0 - 1000. I think the cropping at 255 is just done for convenience, since it is only an 8-bit image and dividing the value by 256 would leave the image with far too less contrast. 

The Noise image is ambient photons (sunlight or other light sources) hitting the detector when the laser is turned off, those photons can be considered as noise.

Chenweng1, 2019-11-26:
Hi @joelsa 

Can you please explain how to interpret the noise value ? Like if one point has higher noise value than another what does that mean ?

joelsa, 2019-11-26:
@Chenweng1 I think the most straightforward approach is to simply interpret the noise data as a camera-like image, as the LiDAR is essentially a low-resolution, large-aperture camera (just another passive-visual sensor) when the laser is turned off, but the detector is turned on. See the following blog post for inspiration on what to use it for: https://ouster.com/blog/the-camera-is-in-the-lidar/
However, I still think that the intensity data is much more useful, as it depends on an intrinsic surface characteristic (i.e. reflectivity), whereas ambient is dependent on external lighting and therefore not really repeatable.

Another approach might be to use the noise data to estimate a reliability measure of the recorded data. This of course requires a solid understand on how variance in measurements varies with the noise level.

dgarbuzov, 2020-12-08:
Only the ROS image topics are currently scaled to 8 bits. For details about how to interpret the data output by the sensor, see the [Software User Manual](https://data.ouster.io/downloads/software-user-manual/software-user-manual-v2p0.pdf#10) section 3.1.2

closed 2020-12-08 by dgarbuzov