ouster-lidar/ouster-sdk #129: SLAM algorithm for Osuter OS1

mrgransky, 2019-11-28:
Hi,

Does anyone know an open source SLAM algorithm already implemented for Ouster Raw data?

Cheers,

bromeara, 2019-12-06:
If you mean the output of this driver then check google cartographer, If you mean raw UDP packets I don't know but I doubt it.

mrgransky, 2020-01-07:
I compiled google cartographer but I do not know if online implementation of this also exist?
Now I first record a rosbag file and run cartographer. What I want is to run both simultaneously!

bromeara, 2020-01-07:
This is more related to cartographer help than ouster help. I'm not sure what you mean run both simultaneously, normally with ros you just run things in multiple terminal instances. There is this page https://google-cartographer-ros.readthedocs.io/en/latest/assets_writer.html but if you want to know more about what cartographer is doing the wiki and the launch files are going to be the best place to look.

mrgransky, 2020-01-10:
What I meant was a single `file.launch` which brings up both _os1_ and _cartographer_ simultaneously and start mapping online in RVIZ as my cart equipped with os1 is moving in the environment!

bromeara, 2020-01-21:
you can just add the os1 code to your cartograther launch file with the xml tags <include file="$(find ouster_ros)/os1.launch> <args name="os1 args" value="os1 values"/> </include> The details of the roslaunch format can be found here http://wiki.ros.org/roslaunch/XML

adityarath302, 2020-03-04:
I tried with Lego-loam .It works but without the IMU just the point clouds.

dgarbuzov, 2020-03-14:
Unfortunately, we currently don't provide a sample SLAM implementation

closed 2020-03-14 by dgarbuzov

YiChenCityU, 2020-07-28:
> I tried with Lego-loam .It works but without the IMU just the point clouds.

Hi, I also use the ouster lidar for lego_loam. However I meet this error about TF. Do you know how to fix it ? Thanks very much.
[image]

adityarath302, 2020-07-29:
Yes I am also getting this issue when i want to vizualize lidar data in RVIZ.Can anyone help regarding this.

bromeara, 2020-09-11:
http://wiki.ros.org/tf#view_frames use this package. You need to ensure there is a connection between base_frame and map