ouster-lidar/ouster-sdk #131: Frame ID names /os1_sensor /os1_imu /os1_lidar

andoreso, 2019-12-01:
Hi, when using the os1.launch file with os1 data it gives the frame IDs names starting with a forwards slash "/", bag files generated this way give the following error when attempting to use with the cartograpther_ros package:

> [FATAL] [1575240562.605128113, 1565427960.543243160]: F1202 00:49:22.000000  9297 sensor_bridge.cc:33] Check failed: frame_id[0] != '/' ('/' vs. '/') The frame_id /os1_lidar should not start with a /. See 1.7 in http://wiki.ros.org/tf2/Migration.

It can be easily fixed by changing the frame ID names in the os1_cloud_node.cpp file. 

Is there a specific reason for the forward slashed?

bromeara, 2019-12-06:
Not besides historical reasons from what I can tell. See PR #56

dgarbuzov, 2020-12-07:
Fixed in #56

closed 2020-12-07 by dgarbuzov