ouster-lidar/ouster-sdk #132: Is there a way to construct a range image directly without going to pointcloud first?

versatran01, 2019-12-07:
I want to construct a range image directly from the packet without converting them to point cloud first.
What I don't understand is that there seems to be an azimuth offset for each beam within a column, so there is no single azimuth angle for each column, rather 64 different ones.
I can't just put the raw range data into the range image because they are acquired at different azimuth angle, right?

versatran01, 2019-12-07:
And also I cannot wrap my head around how does a single firing of the laser generate one column of data, due to the offset (beam_azimuth_angle)

closed 2019-12-08 by versatran01

arunumd, 2023-08-07:
I am interested in an answer to the first question in this thread.

kairenw, 2023-08-07:
Hi @arunumd please open a new ticket for new questions if an issue is already closed, thanks! Feel free to link as needed to closed issues in your new ticket