ouster-lidar/ouster-sdk #137: Failed to initialize sensor

JimmyFroost, 2019-12-16:
Hello,

I am using OS1-16, it was in its bos for a long time but I have to use it again. I used to make it work on a classic Ubuntu, but now I need to run Ubuntu in an Oracle VM virtualBox, from Windows10.

So far, I've been following what's written in Ouster's guide and tutorial.

Now when I apply this line:
`roslaunch src/ouster_ros/os1.launch os1_hostname:=os1.991838000761.local os1_udp_dest:=10.5.5.80 lidar_mode:=1024x20 viz:=true`

I get this message and error:
 
 

> ROS_MASTER_URI=http://localhost:11311
> 
> process[os1_node-1]: started with pid [2791]
> process[os1_cloud_node-2]: started with pid [2792]
> process[viz_node-3]: started with pid [2798]

> [ INFO] [1576510054.932581682]: Connecting to sensor at os1.991838000761.local...
> [ INFO] [1576510054.935488259]: Sending data to 10.5.5.80 using lidar_mode: 1024x20
> getaddrinfo: Name or service not known
> [ERROR] [1576510059.938176847]: Failed to initialize sensor at: os1.991838000761.local
> [ERROR] [1576510059.941696629]: Calling os1 config service failed
> [ERROR] [1576510059.942007566]: Calling os1 config service failed

> 
> ====== ======================================================REQUIRED process [os1_cloud_node-2] has died!
> process has died [pid 2792, exit code 1, cmd /home/julien/Documents/maprog/ouster2/devel/lib/ouster_ros/os1_cloud_node ~/os1_config:=/os1_node/os1_config ~/lidar_packets:=/os1_node/lidar_packets ~/imu_packets:=/os1_node/imu_packets __name:=os1_cloud_node __log:=/home/julien/.ros/log/4e4fc48a-2016-11ea-832a-08002774777a/os1_cloud_node-2.log].
> log file: /home/julien/.ros/log/4e4fc48a-2016-11ea-832a-08002774777a/os1_cloud_node-2*.log
> Initiating shutdown!
> =============== ===============================================

> [viz_node-3] killing on exit
> [os1_cloud_node-2] killing on exit
> [os1_node-1] killing on exit
> shutting down processing monitor...
> ... shutting down processing monitor complete
> done

The dashboard gives me this information

 

> "alerts": {
>     "active": [
>       {
>         "active": true,
>         "category": "INTERNAL_COMM",
>         "cursor": 7,
>         "id": "0x01000013",
>         "level": "WARNING",
>         "msg": "Unit has experienced an internal COMM warning: some measurements may have been skipped.",
>         "msg_verbose": "",
>         "realtime": "6470882511842"
>       },

and 

 

> "log": [
>       {
>         "active": true,
>         "category": "ETHERNET_LINK_BAD",
>         "cursor": 0,
>         "id": "0x01000011",
>         "level": "WARNING",
>         "msg": "Ethernet link bad, please check network switch and harnessing.",
>         "msg_verbose": "Link transitioned to 0/Unknown",
>         "realtime": "281835486671"
>       },

I don't know what to think of the ping answering time :

> 
>  "log": [
>       {
>         "active": true,
>         "category": "ETHERNET_LINK_BAD",
>         "cursor": 0,
>         "id": "0x01000011",
>         "level": "WARNING",
>         "msg": "Ethernet link bad, please check network switch and harnessing.",
>         "msg_verbose": "Link transitioned to 0/Unknown",
>         "realtime": "281835486671"
>       },

Any advice on this issue ? Do you think using a virtual machine could be what's wrong ?
Thanks

mrgransky, 2020-01-14:
Have you tried unplug/plug the network cable and restart your computer? because you don't seem have connection with sensor.

Don't forget:
```
$ cd ~/catkin_ws
$ source devel/setup.bash
```
before running your `roslaunch` command!

fgomezrl87, 2020-01-16:
Hi, 

Any update on this? I have same issue. I am using Raspberry Pi3 and configure a DHCP for Ethernet plug. It is connected to OS1-16, has an ip assigned and it has a response when you ping it. However, after roslaunch, it shows up "Failed to initialize sensor at: 192.168.5.1".

Any help would be appreciated. I already source ros.

JimmyFroost, 2020-01-17:
Hi, 

No good news from me, I tried it 2 days ago again, with the same errors.

fgomezrl87, 2020-01-20:
I found something. In Ouster Software User Guide, you can find that OS1 requieres a network that can provide data throughput of approximately 129 Mbps. Raspberry Pi 3 ethernet works at 100 Mbps. I think that's why it fails. Hopefully someone could confirm this information. In the meantime, i will try using a Raspeberry Pi 4 which has 1Gbps ethernet. I let you know my results.

fgomezrl87, 2020-01-25:
It works for me now. Follow next steps, asumming you are using ubuntu 18, ROS Melodic and Raspberry 

`sudo apt install dnsmasq dhcpcd5
sudo nano /etc/dhcpcd.conf`

Into dhcpcd.conf file, add next lines:

interface eth0
static ip_address=192.168.5.1/24

Restart service

`sudo service dhcpcd restart`

Let's now backup dnsmasq.conf and create a new one.

`sudo mv /etc/dnsmasq.conf /etc/dnsmasq.conf.orig
sudo nano /etc/dnsmasq.conf`

Add next lines to dnsmasq.conf

interface=eth0
dhcp-range=eth0,192.168.5.1,192.168.5.20,255.255.255.0,24h

Notice that eth0 is the ethernet interface on Raspberry Pi. Now make sure OS1 is not plugged in to ethernet interface and run next commands:

`ip addr flush dev eth0
sudo ip addr add 192.168.5.1/24 dev eth0`

Connect os1 and power it on. Then, run:

`sudo ip link set eth0 up
sudo dnsmasq -C /dev/null -kd -F 192.168.5.50,192.168.5.100 -i eth0 --bind-dynamic`

After 15 seconds, you can see DHCP negotiation take place. Now get ip assigned to eth0. 

Use that ip to run Ouster Example. It would be something like this:

`roslaunch os1.launch os1_hostname:=192.168.5.92 os1_udp_dest:=192.168.0.5 lidar_mode:=1024x10 viz:=false`

That's it.

dgarbuzov, 2020-03-14:
`getaddrinfo: Name or service not known` indicates that your system cannot resolve the hostname you're using for the device (`os1.991838000761.local`). Try using the device IP of the device directly.

The `INTERNAL_COMM` alert reported is unrelated, but might indicate an issue with your hardware, and you should contact ouster support to see if that's the case.

closed 2020-03-14 by dgarbuzov

yinwu33, 2021-08-13:
Hello, I met the same error, but in a different situation. It works fine at the beginning, but after I shut down the ROS with `ctrl+c`, it may not be started again unless I unplug/plug the Lidar. This happens not always, and not all lidars. So I'm very confused. Did I shut it down in the wrong way?

kairenw, 2021-08-13:
Hi @yinwu33 -- mind opening another issue instead of having us continue on this one? It'd also be helpful if you could paste in some logs of the error when you try to start it again.

yinwu33, 2021-08-24:
> Hi @yinwu33 -- mind opening another issue instead of having us continue on this one? It'd also be helpful if you could paste in some logs of the error when you try to start it again.

Hello, this issue is solved by updating Firmware. Thanks!