ouster-lidar/ouster-sdk #138: Running two OS-1 on one ethernet driver with ethernet switch

caomuqing, 2019-12-19:
Hi,

I would like to know if there is a standard way to run more than one ouster OS1 lidar on just one Ethernet device, with two lidars connecting to a ethernet switch, which then connects to the ethernet port of my Ubuntu 16.04 computer.

I am able to successfully launch two lidars on one computer(using the os1x2.launch), but I have to assign two separate ethernet connections to them (one lidar connects to the default ethernet port, another connects to an ethernet-usb converter which shows up as a separate ethernet driver in the ifconfig). I have to set up two separate Ethernet connections for them, and hence two lidars are publishing to two  os1_udp_dest addresses. However, this method is troublesome and needs another ethernet-usb converter.

I have tried launching two lidars with one ethernet port, and both of them have the same  os1_udp_dest address, but that makes the nodes not publishing any data at all.

I hope a standard procedure can be provided because this will greatly help users who are using multiple lidars concurrently, which is very common in autonomous vehicle industry.

Best Regards,
Muqing

dgarbuzov, 2020-03-14:
There shouldn't be any issues running two lidars on a single ethernet interface, as long as as the nic and switch support gigabit speeds. Make sure you configure different destination ports for each device.

closed 2020-03-14 by dgarbuzov

caomuqing, 2020-03-16:
Unfortunately this does not solve my issue. I am sure the ethernet switch support gigabit speeds. Can you provide a more detailed setup instruction, such as the launch script, and the network connection(ports) setup?

dgarbuzov, 2020-03-16:
Please contact ouster support for help with multi-sensor configuration. The sample code here doesn't support it out-of-the-box, yet.

dileepamarasinghe, 2020-06-24:
@caomuqing Hello, I am also trying to connect two lidars with the standard port and USB converter to the same computer as you have done. However I can't invoke both at the same time. Only the first one connected works and for the second lidar socket configuration fails. If you have any idea on how to rectify this please let me know. Also, did you write this os1x2.launch file or is it available from somewhere?

ajay1606, 2021-01-07:
Hello, does any have solved this issue? Currently am also trying to connect 2 os1-16 lidars in a single ethernet driver using ethernet switch. But getting an error message like this. Currently both the lidar pointing to the same PORT, if it is port issue, then how could assign a different PORT number?

[image]

dgarbuzov, 2021-01-07:
The error message indicates that you're using the same for multiple nodes. Try updating the `name=` parameter of the relevant `<node>` tags in the launch file

ajay1606, 2021-01-08:
@dgarbuzov Many thanks to you, It worked.

zsun114, 2023-11-18:
> The error message indicates that you're using the same for multiple nodes. Try updating the `name=` parameter of the relevant `<node>` tags in the launch file

I have same issue on this error so how can I update the name= parameter that you mentioned? Thank you!