ouster-lidar/ouster-sdk #143: PTP Grandmaster Clock External Hardware

mrgransky, 2020-01-10:
Suggested by Ouster SW guide, Trimble GM100 is used as an external PTP external grandmaster clock. It seems PTP Grandmaster Clock GM100 has come to an end of its ife product, according to [this](https://www.zti-communications.com/ptp-gm100/).

I was wondering if anyone knows a similar PTP grandmaster clock HW to buy which is most likely research affordable?

Cheers,

daniel-dsouza, 2020-01-11:
If all you need is PTP time stamping, the cheapest solution is to use the hardware you already have - your computer via [ptp4l](https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/6/html/deployment_guide/s1-using_ptp#s2-Checking_for_Driver_and_Hardware_Support)!

mrgransky, 2020-01-11:
I need to synchronize my GPS receiver with my OS1.
Isn’t PTP grandmaster HW provides higher accuracy?

Krishtof-Korda, 2021-01-07:
Hi, @mrgransky 

If you just need to sync GPS with OS1 you can do it via PPS and NMEA over UART, no GM needed. The accuracy will be very much the same as if you did it through PTP and would save you having to purchase an expensive GM.

Just follow the instructions in [How to Synchronize GPS with Ouster Sensor
](https://ouster.atlassian.net/servicedesk/customer/kb/view/988872783) 

Cheers!

closed 2021-04-30 by dgarbuzov