ouster-lidar/ouster-sdk #144: PTP synchronization with LinuxPTP Grandmaster Clock

mrgransky, 2020-01-13:
Sorry for such a long post, in advance!

I have a single NIC on my laptop in which OS1-64 is directly connected and followed the procedure from SW user guide one by one:

1. `$ sudo mkdir -p /etc/systemd/system/ptp4l.service.d`
2. `$ sudo gedit /etc/systemd/system/ptp4l.service.d/override.conf`
3. adding the following lines in `/etc/systemd/system/ptp4l.service.d/override.conf`file:
```
[Service]
ExecStart=
ExecStart=/usr/sbin/ptp4l -f /etc/linuxptp/ptp4l.conf -i enp0s31f6
```
4. Modify the `clockClass` in `/etc/linuxptp/ptp4l.conf`:
#clockClass     248
clockClass      128

5. Start the `ptp4l` service so the change takes effect:
```
$ sudo systemctl daemon-reload
$ sudo systemctl restart ptp4l
$ sudo systemctl status ptp4l

● ptp4l.service - Precision Time Protocol (PTP) service
   Loaded: loaded (/lib/systemd/system/ptp4l.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/ptp4l.service.d
           └─override.conf
   Active: active (running) since Mon 2020-01-13 19:15:48 EET; 31s ago
 Main PID: 6216 (ptp4l)
    Tasks: 1
   Memory: 544.0K
      CPU: 11ms
   CGroup: /system.slice/ptp4l.service
           └─6216 /usr/sbin/ptp4l -f /etc/linuxptp/ptp4l.conf -i enp0s31f6

Jan 13 19:15:48 xenial-ThinkPad-P51 systemd[1]: Started Precision Time Protocol (PTP) service.
Jan 13 19:15:48 xenial-ThinkPad-P51 ptp4l[6216]: [2710.994] selected /dev/ptp0 as PTP clock
Jan 13 19:15:48 xenial-ThinkPad-P51 ptp4l[6216]: [2710.994] port 1: INITIALIZING to LISTENING on INITIALIZE
Jan 13 19:15:48 xenial-ThinkPad-P51 ptp4l[6216]: [2710.994] port 0: INITIALIZING to LISTENING on INITIALIZE
Jan 13 19:15:55 xenial-ThinkPad-P51 ptp4l[6216]: [2717.757] port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Jan 13 19:15:55 xenial-ThinkPad-P51 ptp4l[6216]: [2717.757] selected best master clock 8c1645.fffe.3f002e
Jan 13 19:15:55 xenial-ThinkPad-P51 ptp4l[6216]: [2717.757] assuming the grand master role
```
6. `$ sudo mkdir -p /etc/systemd/system/phc2sys.service.d`
7. `$ sudo gedit /etc/systemd/system/phc2sys.service.d/override.conf` and add following contents:
```
[Service]
ExecStart=
ExecStart=/usr/sbin/phc2sys -w -s CLOCK_REALTIME -c enp0s31f6
```
8. Start the phc2sys service so the change takes effect:
```
$ sudo systemctl daemon-reload
$ sudo systemctl start phc2sys
$ sudo systemctl status phc2sys

● phc2sys.service - Synchronize system clock or PTP hardware clock (PHC)
   Loaded: loaded (/lib/systemd/system/phc2sys.service; disabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/phc2sys.service.d
           └─override.conf
   Active: active (running) since Mon 2020-01-13 19:15:48 EET; 10min ago
 Main PID: 6217 (phc2sys)
   CGroup: /system.slice/phc2sys.service
           └─6217 /usr/sbin/phc2sys -w -s CLOCK_REALTIME -c enp0s31f6

Jan 13 19:26:05 xenial-ThinkPad-P51 phc2sys[6217]: [3328.133] sys offset       344 s2 freq   -3437 delay   4500
Jan 13 19:26:06 xenial-ThinkPad-P51 systemd[1]: Started Synchronize system clock or PTP hardware clock (PHC).
Jan 13 19:26:06 xenial-ThinkPad-P51 phc2sys[6217]: [3329.134] sys offset       231 s2 freq   -3447 delay   4875
Jan 13 19:26:07 xenial-ThinkPad-P51 phc2sys[6217]: [3330.134] sys offset         3 s2 freq   -3605 delay   4625
Jan 13 19:26:08 xenial-ThinkPad-P51 phc2sys[6217]: [3331.134] sys offset      -211 s2 freq   -3818 delay   4625
Jan 13 19:26:09 xenial-ThinkPad-P51 phc2sys[6217]: [3332.134] sys offset      -398 s2 freq   -4069 delay   4625
[... 137 lines cut ...]
    "last_gm_phase_change": "0x0000'0000000000000000.0000",
    "master_offset": 77236,
    "scaled_last_gm_phase_change": 0
  }
}
```
so `parent_data_set.grandmaster_identity` is the identity of the local grandmaster `bc0fa7.fffe.0011bc-1` and `port_data_set.port_state` is `SLAVE`.

16. `$ sudo pmc 'get CURRENT_DATA_SET' -i enp0s31f6`
```
sending: GET CURRENT_DATA_SET
        bc0fa7.fffe.0011bc-1 seq 0 RESPONSE MANAGEMENT CURRENT_DATA_SET 
                stepsRemoved     1
                offsetFromMaster -106139.0
                meanPathDelay    28059.0
```
17. Waiting for 2 min, from `ouster_client` package:
`$ ./ouster_client_example <os1_hostname> <data_destination_ip>`

```
n_lidar_packets     col_0_azimuth          col_0_ts            n_imu_packets           im_av_z           im_la_y                   imu_ts
110553                6.18501          1578938056751576832             17300          0.862122       -0.00952148      1578938056748440843
```
so how can I know my sensor is synchronized successfully? 
Cheers,

krisklau, 2020-04-02:
Hi,

This might be related to your problem, although I have a different setup than yours. (no NTP for instance). 

I noticed like you that I seemed to be getting data offset by 36s, which seemed to be to be way to close to the current [leap-second](https://en.wikipedia.org/wiki/Leap_second) to be a coincidence. 

I adjusted the phc2sys command to force the leap-second parameter to zero like so: 

`ExecStart=/usr/sbin/phc2sys -w -O 0 -s CLOCK_REALTIME -c eth0 -S 1.0`

(The other settings were right for our application, you might want to adjust them. The important for leap-second offset is `-O 0`. More info here: http://manpages.ubuntu.com/manpages/cosmic/man8/phc2sys.8.html)

msadowski, 2020-07-02:
@krisklau many thanks for this solution! You saved me hours of debugging!

rcabg, 2020-10-05:
Hello @krisklau and @mrgransky 

I'm dealing with the exact same problem. Using the commands posted by @mrgransky I get the same results.

If I change just the phc2sys configuration adding -O 0, when I restart everything, chronyc shows a 37 seconds delay, making it unusable by chronyc. 

Is this normal? How did you manage to get a 0 delay with ROS?

Thanks in advance!

Krishtof-Korda, 2021-01-07:
Hi,

This seems to be an issue with Linux PTP not being configured correctly. The sensor will only use the “current_utc_offset”: 37 if ptp4l is transmitting "current_utc_offset_valid": 1. 

If ptp4l transmits "current_utc_offset_valid": 0 the sensor will ignore the offset since it is reported as not valid.

Can you please try the solution mentioned in [this forum post](https://sourceforge.net/p/linuxptp/mailman/message/35955549/)? 

Thanks!

@rcabg, please try `phc2sys -a -r`

closed 2021-04-30 by dgarbuzov

zzodo, 2022-02-10:
Did anyone solved this issue?

I am struggling with the leap second gap in time synchronization with linuxptp.

@Krishtof-Korda 
I have tried to follow the link you commented in (https://github.com/ouster-lidar/ouster_example/issues/204#issuecomment-755799869, https://github.com/ouster-lidar/ouster_example/issues/204#issuecomment-797773875, https://github.com/ouster-lidar/ouster_example/issues/144#issuecomment-755806283)

I really cannot understand why my synchronization still have 37 seconds gap and utcOffsetValid: 0.
I am currently using hard-coded TAI-UTC time offset consideration in driver node.
Can you illustrate some more information to fix my linuxptp configuration in detail?

Thanks in advance.

Durant35, 2022-02-10:
您发送的邮件已经收到。。。

Durant35, 2022-02-11:
As mentioned in http://manpages.ubuntu.com/manpages/jammy/en/man8/phc2sys.8.html
> [image]

And in our project, we use `/lib/systemd/system/phc2sys.service` as following
```bash
[Unit]
Description=Synchronize system clock or PTP hardware clock (PHC)
Documentation=man:phc2sys
Requires=chronyd.service
After=chronyd.service
 
[Service]
Type=simple
# sleep 10s to make sure SYS time synced
ExecStartPre=/bin/sleep 10
## -s device
# Specify the master clock by device (e.g. /dev/ptp0) or interface (e.g. eth0)
#         or by name (e.g. CLOCK_REALTIME for the system clock).
## -c device
# Specify the slave clock by device or interface (e.g. eth1) or by name.
# default: CLOCK_REALTIME
## -O offset
# Specify the offset between the slave and master times in seconds.
## -S step
# Specify the maximum offset (in seconds) the servo corrects by changing the clock
#         frequency instead of stepping the clock.
# default: 0.0, disables stepping after the start.
# 1 (second), robust to GPS-causing system time jump and keep original precision.
ExecStart=/usr/sbin/phc2sys -s CLOCK_REALTIME -c eth3 -O 0 -S 1
 
[Install]
WantedBy=multi-user.target
```

@zzodo Have you tried `-O 0` as above?

Durant35, 2022-02-11:
first, sorry to misleading you, maybe. Actually I never use this LiDAR, and just watched this issue for its general points.

And for your usage, you want to sync your PC(dohoon-NUC8v5PNB) with the OS's lidar via PTP.
- This does not relate to `chrony`, which just used to sync your PC's system time. But recommend you to disable it first to avoid introduce errors due to it. After verified, you can enable it to use.
- For OS's lidar can sync with your PC via PTP, you should first check your system time have synced to the ethernet's PHC, Physical Hardware Clock, you can check it as following. Here `eth2` is the interface name. And `phc2sys` is using to do it.
    ```bash
    sudo phc_ctl eth2 cmp
    phc_ctl[6171.026]: offset from CLOCK_REALTIME is 25ns
    ```
- Then you could check the `ptp4l` status, check the `sudo service ptp4l status` or log as following:
```bash
# cat /var/log/syslog | grep ptp4l
Feb 11 12:41:23 SenseHub ptp4l: [42.058] selected /dev/ptp0 as PTP clock
Feb 11 12:41:23 SenseHub ptp4l: [42.060] port 1: INITIALIZING to LISTENING on INITIALIZE
Feb 11 12:41:23 SenseHub ptp4l: [42.060] port 0: INITIALIZING to LISTENING on INITIALIZE
Feb 11 12:41:23 SenseHub ptp4l: [42.060] port 1: link up
Feb 11 12:41:29 SenseHub ptp4l: [48.125] port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Feb 11 12:41:29 SenseHub ptp4l: [48.126] selected best master clock 6e6691.fffe.d98aa6
Feb 11 12:41:29 SenseHub ptp4l: [48.126] assuming the grand master role
Feb 11 12:47:24 SenseHub ptp4l: [43.049] selected /dev/ptp1 as PTP clock
Feb 11 12:47:24 SenseHub ptp4l: [43.050] port 1: INITIALIZING to LISTENING on INITIALIZE
Feb 11 12:47:24 SenseHub ptp4l: [43.050] port 0: INITIALIZING to LISTENING on INITIALIZE
Feb 11 12:47:24 SenseHub ptp4l: [43.050] port 1: link up
Feb 11 12:47:32 SenseHub ptp4l: [50.126] port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Feb 11 12:47:32 SenseHub ptp4l: [50.126] selected best master clock 3630b8.fffe.c62673
Feb 11 12:47:32 SenseHub ptp4l: [50.126] assuming the grand master role
Feb 11 13:22:30 SenseHub ptp4l: [2148.276] timed out while polling for tx timestamp
Feb 11 13:22:30 SenseHub ptp4l: [2148.276] increasing tx_timestamp_timeout may correct this issue, but it is likely caused by a driver bug
Feb 11 13:22:30 SenseHub ptp4l: [2148.276] port 1: send sync failed
Feb 11 13:22:30 SenseHub ptp4l: [2148.276] port 1: MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
Feb 11 13:22:46 SenseHub ptp4l: [2164.277] port 1: FAULTY to LISTENING on FAULT_CLEARED
Feb 11 13:22:53 SenseHub ptp4l: [2171.268] port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Feb 11 13:22:53 SenseHub ptp4l: [2171.268] selected best master clock 3630b8.fffe.c62673
Feb 11 13:22:53 SenseHub ptp4l: [2171.268] assuming the grand master role
Feb 11 14:11:45 SenseHub ptp4l: [5103.481] timed out while polling for tx timestamp
Feb 11 14:11:45 SenseHub ptp4l: [5103.481] increasing tx_timestamp_timeout may correct this issue, but it is likely caused by a driver bug
Feb 11 14:11:45 SenseHub ptp4l: [5103.481] port 1: send sync failed
Feb 11 14:11:45 SenseHub ptp4l: [5103.481] port 1: MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
Feb 11 14:12:01 SenseHub ptp4l: [5119.482] port 1: FAULTY to LISTENING on FAULT_CLEARED
Feb 11 14:12:08 SenseHub ptp4l: [5126.305] port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
Feb 11 14:12:08 SenseHub ptp4l: [5126.305] selected best master clock 3630b8.fffe.c62673
Feb 11 14:12:08 SenseHub ptp4l: [5126.305] assuming the grand master role
```
- Last but not least, check the LiDAR's status, like HESAI's PandarQT, we check it as following:
[image]

If all above work fine, then it's related to the Driver, check the timestamp in every lidar packet?

zzodo, 2022-02-11:
Thank you for your detailed instruction!

And yes, you got my point.
My aim for this synchronization procedure is to get system-timestamped data from Ouster lidar.

- I really did not recognize that chrony does not related to the synchronization within my application. Thank you for pointing the exact usage.

- The synchronization between PHC and system time is shown as below.
```
dohoon@dohoon-NUC8v5PNB:~$ sudo phc_ctl eno1 cmp
phc_ctl[6486.651]: offset from CLOCK_REALTIME is -885ns
```
- And the status of `ptp4l` is:
```
dohoon@dohoon-NUC8v5PNB:~$ sudo service ptp4l status
● ptp4l.service - Precision Time Protocol (PTP) service
     Loaded: loaded (/lib/systemd/system/ptp4l.service; enabled; vendor preset: enabled)
    Drop-In: /etc/systemd/system/ptp4l.service.d
             └─override.conf
     Active: active (running) since Fri 2022-02-11 14:24:36 KST; 1h 36min ago
       Docs: man:ptp4l
   Main PID: 3241 (ptp4l)
      Tasks: 1 (limit: 38069)
     Memory: 264.0K
     CGroup: /system.slice/ptp4l.service
             └─3241 /usr/sbin/ptp4l -f /etc/linuxptp/ptp4l.conf

 2월 11 14:35:44 dohoon-NUC8v5PNB ptp4l[3241]: [1452.361] port 1: link down
 2월 11 14:35:44 dohoon-NUC8v5PNB ptp4l[3241]: [1452.361] port 1: LISTENING to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
 2월 11 14:35:44 dohoon-NUC8v5PNB ptp4l[3241]: [1452.361] selected local clock a4ae11.fffe.1ca3b0 as best master
 2월 11 14:35:44 dohoon-NUC8v5PNB ptp4l[3241]: [1452.361] assuming the grand master role
 2월 11 14:35:48 dohoon-NUC8v5PNB ptp4l[3241]: [1455.959] port 1: link up
 2월 11 14:35:48 dohoon-NUC8v5PNB ptp4l[3241]: [1455.959] port 1: FAULTY to LISTENING on INIT_COMPLETE
 2월 11 14:35:55 dohoon-NUC8v5PNB ptp4l[3241]: [1462.770] port 1: LISTENING to MASTER on ANNOUNCE_RECEIPT_TIMEOUT_EXPIRES
 2월 11 14:35:55 dohoon-NUC8v5PNB ptp4l[3241]: [1462.770] selected local clock a4ae11.fffe.1ca3b0 as best master
 2월 11 14:35:55 dohoon-NUC8v5PNB ptp4l[3241]: [1462.770] assuming the grand master role
```
- For the device status, response from Ouster lidar shows like:
```
dohoon@dohoon-NUC8v5PNB:~$ http os1-991949000668.local/api/v1/time/ptp
HTTP/1.1 200 OK
content-length: 1314
content-type: application/json

{
    "current_data_set": {
        "mean_path_delay": 38775.0,
        "offset_from_master": -11895.0,
        "steps_removed": 1
    },
    "parent_data_set": {
        "gm_clock_accuracy": 254,
        "gm_clock_class": 128,
        "gm_offset_scaled_log_variance": 65535,
        "grandmaster_identity": "a4ae11.fffe.1ca3b0",
        "grandmaster_priority1": 128,
        "grandmaster_priority2": 128,
        "observed_parent_clock_phase_change_rate": 2147483647,
        "observed_parent_offset_scaled_log_variance": 65535,
        "parent_port_identity": "a4ae11.fffe.1ca3b0-1",
        "parent_stats": 0
    },
    "port_data_set": {
        "announce_receipt_timeout": 3,
        "delay_mechanism": 1,
        "log_announce_interval": 1,
        "log_min_delay_req_interval": 0,
        "log_min_pdelay_req_interval": 0,
        "log_sync_interval": 0,
        "peer_mean_path_delay": 0,
        "port_identity": "bc0fa7.fffe.001535-1",
        "port_state": "SLAVE",
        "version_number": 2
    },
    "profile": "default",
    "time_properties_data_set": {
        "current_utc_offset": 37,
        "current_utc_offset_valid": 1,
        "frequency_traceable": 0,
        "leap59": 0,
        "leap61": 0,
        "ptp_timescale": 1,
        "time_source": 160,
        "time_traceable": 1
    },
    "time_status_np": {
        "cumulative_scaled_rate_offset": 0.0,
        "gm_identity": "a4ae11.fffe.1ca3b0",
        "gm_present": true,
        "gm_time_base_indicator": 0,
        "ingress_time": 1644563028927961646,
        "last_gm_phase_change": "0x0000'0000000000000000.0000",
        "master_offset": -11895,
        "scaled_last_gm_phase_change": 0
    }
}
```
which seems appropriate to me.

Maybe I should debug the raw timestamp transferred from the sensor, as you suggested.
Thank you again for kind help!

Krishtof-Korda, 2022-02-11:
Hi,

You have correctly synchronized the Ouster!

The Ouster sensor always timestamps with the master time which for PTP is
TAI time without utc corrections. You would need to add the master reported
UTC offset to the Ouster timestamp in your code.

Cheers,

Krishtof
Field Application Engineer
ouster.com
775.745.4045
calendly.com/krishtof-korda