ouster-lidar/ouster-sdk #145: Empty viz window

jattiogbe, 2020-01-15:
Hello,

I haven't been using the OS1-16 for a while, now updated to V1.13, I built latest sources.

When typing:
`roslaunch src/ouster_ros/os1.launch os1_hostname:=os1-991....local os1_udp_dest:=10.5.5.20 lidar_mode:=1024x10 viz:=true`

Everything seems fine :

started roslaunch server http://julienBoost:45241/

```
SUMMARY
========

PARAMETERS
 * /os1_node/lidar_mode: 1024x10
 * /os1_node/metadata: 
 * /os1_node/os1_hostname: os1-991838000761....
 * /os1_node/os1_imu_port: 7503
 * /os1_node/os1_lidar_port: 7502
 * /os1_node/os1_udp_dest: 10.5.5.20
 * /os1_node/replay: False
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    os1_cloud_node (ouster_ros/os1_cloud_node)
    os1_node (ouster_ros/os1_node)
    viz_node (ouster_ros/viz_node)

auto-starting new master
process[master]: started with pid [25806]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to c8a1a74c-377e-11ea-a23d-0c9d9256d8d0
process[rosout-1]: started with pid [25817]
started core service [/rosout]
process[os1_node-2]: started with pid [25823]
process[os1_cloud_node-3]: started with pid [25825]
process[viz_node-4]: started with pid [25831]
[ INFO] [1579082826.660921728]: Connecting to sensor at os1-991838000761.local...
[ INFO] [1579082826.662285981]: Sending data to 10.5.5.20 using lidar_mode: 1024x10
[ INFO] [1579082826.670426076]: Sensor reconfigured successfully, waiting for data...
[ INFO] [1579082826.670722765]: Wrote metadata to os1-991838000761.local.dashbo[ INFO] [1579082826.670923326]: Sensor sn: 991838000761 firmware rev: v1.13.0
```

BUT the Visualisation toolkit window appears and remains black and empty.

In the dashboard, I have this strange information about a bad ethernet link whereas I am able to display live data with OusterStudio :

> 
> {
>   "alerts": {
>     "active": [
>       {
>         "active": true,
>         "category": "UDP_CONNECT",
>         "cursor": 18,
>         "id": "0x01000033",
>         "level": "WARNING",
>         "msg": "Couldn't resolve IP address; check network and udp_ip.",
>         "msg_verbose": "Failed to open lidar UDP socket with destination host 10.5.5.20:7502:7502",
>         "realtime": "4591968685590"
>       },
>       {
>         "active": true,
>         "category": "UDP_CONNECT",
>         "cursor": 19,
[... 223 lines cut ...]
>     "sync_pulse_in_polarity": "ACTIVE_HIGH",
>     "sync_pulse_out_angle": 360,
>     "sync_pulse_out_frequency": 1,
>     "sync_pulse_out_polarity": "ACTIVE_HIGH",
>     "sync_pulse_out_pulse_width": 10,
>     "timestamp_mode": "TIME_FROM_INTERNAL_OSC",
>     "udp_ip": "10.5.5.20:7502",
>     "udp_port_imu": 7503,
>     "udp_port_lidar": 7502
>   },
>   "info": {
>     "base_pn": "000-101323-03",
>     "base_sn": "101836000098",
>     "build_date": "2019-11-04T23:58:36Z",
>     "build_rev": "v1.13.0",
>     "image_rev": "ousteros-image-prod-aries-v1.13.0-20191105025459",
>     "prod_line": "OS-1-16-A0",
>     "prod_pn": "840-101396-03",
>     "prod_sn": "991838000761",
>     "proto_rev": "v1.1.1",
>     "status": "RUNNING"
>   }
> }

Can anyone help me ?

zuzi-m, 2020-01-20:
it seems like the `udp_ip` is being set incorrectly - it is including the port in the IP (see the status you included - it says `"udp_ip": "10.5.5.20:7502",` and `"udp_port_lidar": 7502` and the error messages say that the lidar is unable to open a socket with host `10.5.5.20:7502:7502` which is obviously not a valid host)

Can you make sure that you are running the roslaunch file with correct parameters? (especially the `os1_udp_dest`)

Also, to validate that the other nodes running from the roslaunch file are running correctly, you can have a look if they are publishing the `/os1_node/lidar_packets` and `/os1_cloud_node/points` topics?

jattiogbe, 2020-01-21:
Hello,

Thanks for pointing that out, it was my mistake during one try indeed.
When corrected, the issue is still there though...

Even when my command is 
`
roslaunch src/ouster_ros/os1.launch os1_hostname:=os1-991838000761.local os1_udp_dest:=10.5.5.20 lidar_mode:=1024x10 viz:=true`
the diagnostic line in return is :
> 
"msg_verbose": "Failed to open lidar UDP socket with destination host 192.168.1.7:7502",

I don't understand why both adresses are different.

I tried  **rostopic echo /os1_node/lidar_packets**  and  **rostopic echo /os1_cloud_node/points**, with no data in the output...

zuzi-m, 2020-01-21:
In that case it seems like the roslaunch and all the ROS nodes are completely unable to get any data because the lidar unit is unable to successfully start the connection. (you can try running `rostopic list` to see if there are any topics that could be from ouster_ros package and see if they publish some data)

However, the `UDP_CONNECT` and `ETHERNET_LINK_BAD` errors might not be the actual cause of the problems you have. They might just be there because at some point the connection was not working (e.g. when you powered up the lidar unit but no destination udp ip was set in the lidar, or the udp ip was set already but no program was listening on that IP and port). These warnings will appear in some cases but will not cause the lidar to not work once everything is running.

What seems to be a problem are the `INTERNAL_FAULT` errors. I don't know how to deal with them because they don't have a very helpful description and I have never encountered them myself.

Have you tried connecting to your lidar unit using the TCP connection? Does that work normally? If yes, maybe the problem can be fixed by reinitializing the lidar using the TCP connection, or changing some settings.
Also, have you tried resetting the lidar's config to default using the web interface?
You said that you are able to feed the lidar data to Ouster Studio - try comparing the output of lidar's diagnostics from the dashboard when you are connected to ouster studio vs when you are trying to use the ROS package.

dgarbuzov, 2020-03-15:
The `INTERNAL_FAULT` errors could indicate hardware issues and you should contact ouster support.

The `UDP_CONNECT` warnings probably just mean that you have a different udp destination address persisted in the configuration. On startup, the unit detected that it couldn't reach that address before the configuration was updated by the client. If you save a new configuration with `write_config_txt`, these should go away.

closed 2020-03-15 by dgarbuzov