ouster-lidar/ouster-sdk #146: decode /os1_node/imu_packets and /os1_node/lidar_packets

mrgransky, 2020-01-17:
Given a `file.bag` with following info: `$ rosbag info file.bag --freq`

```
path:        file.bag
version:     2.0
duration:    1:04s (64s)
start:       May 07 2019 21:39:57.03 (1557254397.03)
end:         May 07 2019 21:41:01.58 (1557254461.58)
size:        500.1 MB
messages:    47768
compression: none [656/656 chunks]
types:       ouster_ros/PacketMsg [4f7b5949e76f86d01e96b0e33ba9b5e3]
topics:      /os1_node/imu_packets      6456 msgs    : ouster_ros/PacketMsg
             /os1_node/lidar_packets   41312 msgs    : ouster_ros/PacketMsg
```
I would like to decode it into: `/os1_cloud_node/imu` and  `/os1_cloud_node/points` with the following python script:

```
import rospy
from ouster_ros.msg import PacketMsg

def cb(msg):
    rospy.loginfo('msg: '.format(msg.buf)) <<--------- How/What to read/extract??

def main():
    rospy.init_node("my_sub")
    rospy.Subscriber('/os1_node/imu_packets', PacketMsg,cb)
    rospy.spin()

if __name__ == "__main__":
    main()
```
The PacketMsg in ouster_ros, however, is
```
$ rosmsg show ouster_ros/PacketMsg
uint8[] buf
```
Can anyone help me on the saved data in `uint8[] buf` ? Is there any documentation available?
PS. I have read Software Manual in [here](https://data.ouster.io/downloads/v1.12.0-sw-user-guide.pdf)!

zuzi-m, 2020-01-20:
The data in `uint8[] buf` is binary data coming from the OS-1 sensor. It has exactly the same format as the data described in the [software used guide](https://data.ouster.io/downloads/software-user-guide-v1.13.0.pdf), chapter 3.4 Lidar Data Format for `lidar_packets` topic and chapter 3.5 IMU Data Format for `imu_packets`.

These ROS messages are just wrapping the packets coming from the network so that instead of listening to the network packets directly, you can use ROS and listen to the `/os1_node/imu_packets` and `/os1_node/lidar_packets` topics to get the data from the sensor, and possibly save the data as a ROS bag file.

To make these topics useful, you dont have to read and decode them manually because the `ouster_ros` package has a node `os1_cloud_node` which will listen to the packets from those two topics, decode them, and publishes the cloud and the IMU messages in ROS topics.

zuzi-m, 2020-01-22:
@mrgransky you mentioned this issue yesterday but have not reacted here. Does my reply answer your question? Why do you need to decode the binary data into the point cloud and imu topics yourself? Can you use the `os1_cloud_node` to do it for you?

TaylorDale, 2020-01-28:
>  Why do you need to decode the binary data into the point cloud and imu topics yourself? Can you use the `os1_cloud_node` to do it for you?

Perhaps to account for movement of the lidar unit between scanlines which os1_cloud_node wouldn't account for?

zuzi-m, 2020-01-28:
@TaylorDale good point, that would make sense

So in terms of this issue, has the question been sufficiently answered by what I said earlier?

Another thing that comes to my mind - if anyone wants to see an example of how to decode the binary data in the `/os1_node/lidar_packets` and `/os1_node/imu_packets` topics, just have a look at the source code of the [os1_cloud_node](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/os1_cloud_node.cpp), especially the way the [lidar data is read](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/src/os1_cloud_node.cpp#L57-L63), which comes down to [this function](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/os1_util.h#L91-L140) where `packet_buf` is the pointer to the `uint8[] buf` from the `ouster_ros/PacketMsg` message type mentioned in the original question. The function takes the data of one UDP packet from the lidar and decodes it into 16 columns of 64 points, as per section 3.4 Lidar Data Format from the software user guide.

The code is a bit difficult to read because it uses iterators and lambda functions, but after some time it becomes clear, especially if you studied the software manual. It can be easily adjusted to read the data directly from a network socket or batch the data differently than using an iterator.

dgarbuzov, 2020-03-14:
The relevant code is in [batch_to_iter](https://github.com/ouster-lidar/ouster_example/blob/1cf6faf3b993caab483b7e9b25a0f287900acc2a/ouster_client/include/ouster/os1_util.h#L83)

See [os1_cloud_node](https://github.com/ouster-lidar/ouster_example/blob/1cf6faf3b993caab483b7e9b25a0f287900acc2a/ouster_ros/src/os1_cloud_node.cpp#L59) for an example of how it's used.

closed 2020-03-14 by dgarbuzov