ouster-lidar/ouster-sdk #148: TF Pose

pcarrascof, 2020-01-27:
I would like to know the exact reference between the transforms: 'os1_sensor', 'imu_sensor' and 'lidar sensor' in the real device.

zuzi-m, 2020-01-29:
What do you mean the exact reference between the transforms?

The transformations themselves capture the relative position between the frames, and it is all published by the `os1_cloud_node`.

Moreover, the exact position between the frames are described in the [software user guide](https://data.ouster.io/downloads/software-user-guide-v1.13.0.pdf) in chapter **4 Coordinate Frames**

pcarrascof, 2020-02-04:
Hello,

The doubt is related to the transforms in the real device. I understand that the node provided shows the relationships between os1_imu, and os1_lidar to os1_sensor. 
The question that remains unsolved and that I do not find in the software guide is the exact location of this last transform in the device ('os1_sensor'). 
I do not know if it is located in the middle of the device, referring to this at half height on the z-axis.

To be even clearer, the adjudicated relationship is provided in the images 'provided_transform.png and 'provided_transform_2.png'.
But not the location of the 'os1_sensor' transform, which could be anywhere on the real device as the other image ('tf_os1_sensor.png') shows.

Thank you for your time,
Regards.

Paloma Carrasco Fernández
Perception & Software Unit
Advanced Center for Aerospace Technologies (CATEC)
Parque Tecnológico y Aeronáutico de Andalucía
C/ Wilbur y Orville Wright 19 - 41309 La Rinconada, Sevilla (SPAIN)
Phone: +34 954 179 002
www.catec.aero
www.atlascenter.aero
Quality Certifications - Disclaimer

zuzi-m, 2020-02-04:
If I understand your question correctly, you are not sure where on the physical device the origin of the `os1_sensor` coordinate frame is, vertically (on Z axis).

That is explained on the first paragraph of chapter 4 in the software guide:

> The Sensor Coordinate Frame follows the right-hand rule convention and is defined at the center of the sensor housing **on the bottom**, with the x-axis pointed forward, y-axis pointed to the left and z-axis pointed towards the top of the sensor. The external connector is located in the negative x direction.

So, the origin of the `os1_sensor` coordinate frame is on the bottom of the actual lidar unit (without the removable heat sink plate). It is true that the software user guide could say this more clearly and include a side/front view of the lidar unit to show the origins better.

Btw, it seems like you included some images in your reply that you mention, but they didn't seem to go through, so please let me know if I understood your question and answered it sufficiently. If not, please try uploading the pictures and clarify further.

pcarrascof, 2020-02-13:
I would like to know if it can be possible that you can draw in the picture the height of the Z axis.
Because I can't solve my doubt. 
It has been link the image using WeTransfer (https://we.tl/t-zEJqhnxA19).

zuzi-m, 2020-02-13:
It actually already is in the picture that you linked - you can notice the zeros in the image. They correspond to the origin of the `os1_sensor` coordinate frame.

I marked the origin of `os1_sensor` coordinate frame by red color in the image. Blue color in the image shows the position of the `os1_lidar` coordinate frame on the lidar unit. Hope this gives a clear explanation of where the origins are and why the transformation between them is the way it is (e.g. it includes a translation by 36.18mm along the Z axis).
[image]

closed 2020-12-07 by dgarbuzov