ouster-lidar/ouster-sdk #149: Image saving from Ouster LiDAR

ethan20120, 2020-01-29:
I am trying to save the intensity image from the rosbag into the .jpg format. However, the image size is so small that it is very difficult to understand the image. I am using the "rosrun image_view image_saver image:=/img_node/intensity_image " command to save my image.
Hi Altruists,

Can you please suggest me whether there is any possibility to getter image and in a large size from the Ouster Sensor's image node (range,ambient, intensity)? If so, can you please guide me the process to extract the image from the ros bag in a better way.

I have attached some images following my procedure

[image]

[image]
[image]

Thank you very much fro your valuable time. 

Best

zuzi-m, 2020-01-29:
The image you get is produced from the lidar data that you are getting, the image is produced from the points of the point cloud you get from the lidar and therefore it has the same dimensions of the point cloud,

The resolution of the image is always 64 pixels in height (because OS-1 lidar produces 64 points per column) and the width depends on the mode you use, which is either 512, 1024, or 2048 pixels.

What you can do is enlarge the image (obviously, but it will not look any better because it will be pixelated) or interpolate the values that you get from your  point cloud and produce the image yourself (e.g. if your lidar unit uses only 16 or 32 channels, you will have "empty rows" in your image and you can fill these gaps in by interpolating the values between rows with actual data).

There are many ways to do the interpolation but this repository is not meant to do this (as far as the code is now) so I believe it is not an actual "issue" to solve here.

dgarbuzov, 2020-12-07:
Don't have any plans to publish pre-scaled images. You should be able to post-process the raw images using e.g. [imagemagick](https://imagemagick.org/index.php)'s `convert` tool

closed 2020-12-07 by dgarbuzov