ouster-lidar/ouster-sdk #154: Timestamp is 24 hours in the future when using pulse sync with NMEA

msadowski, 2020-03-26:
Hi!

I don't think it's strictly related to the ROS driver, however I think some of you might have figured this out. Here is my time info:

```
get_time_info
{"timestamp": {"time": 1585326910.670651, "mode": "TIME_FROM_SYNC_PULSE_IN", 
"time_options": {"ptp_1588": 181, "sync_pulse_in": 1585326910, "internal_osc": 172}}, 
"sync_pulse_in": {"locked": 1, "polarity": "ACTIVE_LOW", "diagnostics": {"last_period_nsec": 0, 
"count": 173, "count_unfiltered": 1067036}}, "multipurpose_io": {"mode": "INPUT_NMEA_UART", 
"sync_pulse_out": {"polarity": "ACTIVE_HIGH", "frequency_hz": 1,
"angle_deg": 360, "pulse_width_ms": 10}, "nmea": {"locked": 1, "polarity": "ACTIVE_HIGH", 
"ignore_valid_char": 0, "baud_rate": "BAUD_115200", "leap_seconds": 0, "diagnostics": {"decoding": 
{"utc_decoded_count": 171, "date_decoded_count": 170, "not_valid_count": 0, "last_read_message": 
"GPRMC,163509.00,A,3154.13676,N,09324.17018,W,0.021,,260320,,,R,V*07"}, "io_checks": 
{"start_char_count": 171, "char_count": 11968, "bit_count": 35345, "bit_count_unfilterd": 
17614729}}}}}
```

The time is equal to 1585326910 (GMT: Friday, March 27, 2020 4:35:10 PM)

The NMEA message: "$GPRMC,163509.00,A,3154.13676,N,09324.17018,W,0.021,,260320,,,R,V*07" (the checksum will be wrong as I modified the lat/lon) reports a time of Thu, 26 Mar 2020 16:35:09 UTC (according to https://rl.se/gprmc)

Did anyone stumble upon that? Is there anything I'm missing here? It's odd that the offset is exactly 24 hours.

cs23, 2020-03-26:
Can confirm, this was from earlier today (GPRMC shows 260320), yet UNIX time shows being March 27, 2020.
[image]

Howonjung, 2020-04-23:
Did you find solution for that since I have the exactly same symptoms as you had ?

msadowski, 2020-04-23:
For the time being I manually subtract the 24 hour offset in the ROS driver like this: https://github.com/msadowski/ouster_example/commit/5c72466ec767fd8301c9bb53f2dcaf4a68645e42 . Didn't do extensive testing yet but it should do the job until fix is released.

TaylorDixonHD, 2020-05-28:
can confirm with 2 seperate units and models 64&32:

 {"timestamp": {"time": 1590789747.731945, "mode": "TIME_FROM_SYNC_PULSE_IN", "time_options": {"ptp_1588": 357, "sync_pulse_in": 1590789745, "internal_osc": 349}}, "sync_pulse_in": {"locked": 0, "polarity": "ACTIVE_HIGH", "diagnostics": {"last_period_nsec": 0, "count": 148, "count_unfiltered": 41923}}, "multipurpose_io": {"mode": "INPUT_NMEA_UART", "sync_pulse_out": {"polarity": "ACTIVE_HIGH", "frequency_hz": 1, "angle_deg": 360, "pulse_width_ms": 10}, "nmea": {"locked": 1, "polarity": "ACTIVE_LOW", "ignore_valid_char": 0, "baud_rate": "BAUD_9600", "leap_seconds": 0, "diagnostics": {"decoding": {"utc_decoded_count": 44, "date_decoded_count": 43, "not_valid_count": 0, "last_read_message": "GPRMC,220231.00,A,3216.66279,N,09516.60365,W,0.138,233.3,280520,2.1,E,D*20"}, "io_checks": {"start_char_count": 44, "char_count": 3332, "bit_count": 9825, "bit_count_unfilterd": 9824}}}}}

{"timestamp": {"time": 1590788314.495152, "mode": "TIME_FROM_SYNC_PULSE_IN", "time_options": {"ptp_1588": 309, "sync_pulse_in": 1590788314, "internal_osc": 302}}, "sync_pulse_in": {"locked": 1, "polarity": "ACTIVE_HIGH", "diagnostics": {"last_period_nsec": 0, "count": 302, "count_unfiltered": 250}}, "multipurpose_io": {"mode": "INPUT_NMEA_UART", "sync_pulse_out": {"polarity": "ACTIVE_HIGH", "frequency_hz": 1, "angle_deg": 360, "pulse_width_ms": 10}, "nmea": {"locked": 1, "polarity": "ACTIVE_LOW", "ignore_valid_char": 0, "baud_rate": "BAUD_9600", "leap_seconds": 0, "diagnostics": {"decoding": {"utc_decoded_count": 301, "date_decoded_count": 301, "not_valid_count": 0, "last_read_message": "GPRMC,213834.00,A,3216.66399,N,09516.60097,W,0.312,280.8,280520,2.1,E,D*27"}, "io_checks": {"start_char_count": 301, "char_count": 23042, "bit_count": 68200, "bit_count_unfilterd": 68199}}}}}

this post gave me back some sanity points thanks.

TaylorDale, 2020-06-19:
We've also run into this issue - thanks for the fix.

TaylorDale, 2020-06-20:
We've set the leap seconds parameter to a day to fix the issue without modifying the driver

Krishtof-Korda, 2020-12-21:
Hi!

This bug has been **fixed** in the latest release of the firmware for Ouster sensors, [**FW v2.0**](https://ouster.com/downloads).

Please read the blog post ["Firmware v2.0 optimizes data processing and boosts sensor performance for all sensors"](https://ouster.com/blog/firmware-v2.0-optimizes-data-processing-and-boosts-sensor-performance-for-all-sensors/) to get familar. 

You can download FW v2.0 at the **[ouster.com/downloads](https://ouster.com/downloads)** page, where you can also find the latest User Manuals, Changelogs, CAD models, etc. 

This fix is noted in the Changelog as:

- Corrected NMEA message decoding (date calculation on leap years).

Thanks!

closed 2020-12-21 by dgarbuzov