ouster-lidar/ouster-sdk #165: OS1-64 Gen 1 Curved / Concave Surfaces at Near Distances in Point Clouds

pauldeee, 2020-05-25:
Hi all,

Recently got my hands on a OS1-64 Gen1 sensor. I upgraded the firmware from v1.11 directly to v1.13. I am currently notcing that at short distances, about 3m or less, that there is a noticable amount of concave distortion to the point cloud... Which leads me to have some questions:

1) Might this be a ROS Driver issue and do I just have an incorrect setting somewhere?

2) Could this be an issue with the hardware, possibly a manufacturing defect?

3) Could this be the inherent behavior of this sensor, to have distortion like this? (I'd hope not)

4) Might there be some calibration process I've missed or don't know about?

5) Has anyone else has noticed this or has seen anything similar?

I've attached a couple screenshots. First screenshot is of a interior wall inside a house. The Door way and nearby wall clearly bowed. Second screenshot is of a different view, for reference the axes are length of 1m and I've circled the area I am referencing. The house walls are not warped like this.

[image]
[image]

Here is a ICP based map of the interior on which I made a couple annotations, concave / bowed walls are quite easy to see: https://sketchfab.com/3d-models/ouster-test-73274d50d5c544c99aef39810fc02379

Note: I've reached out to Ouster about this about 1 week ago, but I've yet to hear back so I thought I'd ask around here in the meantime.

rcabg, 2020-11-12:
Hi @pauldeee 

I have exactly the same problem with the same sensor. Firmware in my case is 1.14-beta-11. Did you solve this issue?

Thanks in advance!

pauldeee, 2020-11-12:
Hi @rcabg,

Unfortunately this has not yet been resolved for me. I've been talking with Ouster support for the better part of 6 months now. I can no longer say that I am hopeful that there will be a resolution. I've updated to the newest beta and that did not help.

I'm hoping for better luck on your end. If you come across something please let me know.

closed 2021-03-17 by pauldeee

themightyoarfish, 2024-06-11:
The support once stated that OS0 does have some barrel distortion at near range, though obviously this should be publicly documented. I've heard rumors that newer firmwares (likely only rev7 sensors) exhibit this problem to a lesser extent.