ouster-lidar/ouster-sdk #166: sync_pulse_in intermittent lock

dmbr0, 2020-05-28:
GPS IMU unit is outputting RMC sentence and PPS info correctly. The lidar unit is intermittently locking onto the sync_pulse_in.

Below are the settings I'm using and output from get_time_info showing the intermittent locking.

I don't know if this is a bug or I'm just doing something wrong. Any help would be appreciated. 

set_config_param multipurpose_io_mode INPUT_NMEA_UART 
set_config_param nmea_in_polarity ACTIVE_LOW 
set_config_param nmea_baud_rate BAUD_9600 
set_config_param nmea_leap_seconds 27
set_config_param nmea_ignore_valid_char 1
set_config_param sync_pulse_out_frequency 1
set_config_param timestamp_mode TIME_FROM_SYNC_PULSE_IN
set_config_param sync_pulse_in_polarity ACTIVE_LOW
reinitialize
write_config_txt

{"timestamp": {"time": 1590636495.041678, "mode": "TIME_FROM_SYNC_PULSE_IN", "time_options": {"ptp_1588": 3928, "sync_pulse_in": 1590636491, "internal_osc": 3921}}, "sync_pulse_in": {"locked": 0, "polarity": "ACTIVE_LOW", "diagnostics": {"last_period_nsec": 0, "count": 3, "count_unfiltered": 1}}, "multipurpose_io": {"mode": "INPUT_NMEA_UART", "sync_pulse_out": {"polarity": "ACTIVE_HIGH", "frequency_hz": 1, "angle_deg": 360, "pulse_width_ms": 10}, "nmea": {"locked": 1, "polarity": "ACTIVE_LOW", "ignore_valid_char": 1, "baud_rate": "BAUD_9600", "leap_seconds": 27, "diagnostics": {"decoding": {"utc_decoded_count": 3494, "date_decoded_count": 3493, "not_valid_count": 1, "last_read_message": "GPRMC,032813.00,A,4013.03552,N,08308.98462,W,0.0,168.7,280520,,,d*62"}, "io_checks": {"start_char_count": 3555, "char_count": 250990, "bit_count": 742302, "bit_count_unfilterd": 742301}}}}}

get_time_info
{"timestamp": {"time": 1590636491.293353, "mode": "TIME_FROM_SYNC_PULSE_IN", "time_options": {"ptp_1588": 3928, "sync_pulse_in": 1590636491, "internal_osc": 3922}}, "sync_pulse_in": {"locked": 1, "polarity": "ACTIVE_LOW", "diagnostics": {"last_period_nsec": 0, "count": 3, "count_unfiltered": 1}}, "multipurpose_io": {"mode": "INPUT_NMEA_UART", "sync_pulse_out": {"polarity": "ACTIVE_HIGH", "frequency_hz": 1, "angle_deg": 360, "pulse_width_ms": 10}, "nmea": {"locked": 1, "polarity": "ACTIVE_LOW", "ignore_valid_char": 1, "baud_rate": "BAUD_9600", "leap_seconds": 27, "diagnostics": {"decoding": {"utc_decoded_count": 3494, "date_decoded_count": 3493, "not_valid_count": 1, "last_read_message": "GPRMC,032813.00,A,4013.03552,N,08308.98462,W,0.0,168.7,280520,,,d*62"}, "io_checks": {"start_char_count": 3555, "char_count": 250990, "bit_count": 742302, "bit_count_unfilterd": 742301}}}}}

get_time_info
{"timestamp": {"time": 1590636491.825587, "mode": "TIME_FROM_SYNC_PULSE_IN", "time_options": {"ptp_1588": 3929, "sync_pulse_in": 1590636491, "internal_osc": 3922}}, "sync_pulse_in": {"locked": 1, "polarity": "ACTIVE_LOW", "diagnostics": {"last_period_nsec": 0, "count": 3, "count_unfiltered": 1}}, "multipurpose_io": {"mode": "INPUT_NMEA_UART", "sync_pulse_out": {"polarity": "ACTIVE_HIGH", "frequency_hz": 1, "angle_deg": 360, "pulse_width_ms": 10}, "nmea": {"locked": 1, "polarity": "ACTIVE_LOW", "ignore_valid_char": 1, "baud_rate": "BAUD_9600", "leap_seconds": 27, "diagnostics": {"decoding": {"utc_decoded_count": 3495, "date_decoded_count": 3494, "not_valid_count": 1, "last_read_message": "GPRMC,032814.00,A,4013.03552,N,08308.98462,W,0.0,168.7,280520,,,d*65"}, "io_checks": {"start_char_count": 3556, "char_count": 251061, "bit_count": 742506, "bit_count_unfilterd": 742505}}}}}

get_time_info
{"timestamp": {"time": 1590636492.410455, "mode": "TIME_FROM_SYNC_PULSE_IN", "time_options": {"ptp_1588": 3930, "sync_pulse_in": 1590636491, "internal_osc": 3923}}, "sync_pulse_in": {"locked": 0, "polarity": "ACTIVE_LOW", "diagnostics": {"last_period_nsec": 0, "count": 3, "count_unfiltered": 1}}, "multipurpose_io": {"mode": "INPUT_NMEA_UART", "sync_pulse_out": {"polarity": "ACTIVE_HIGH", "frequency_hz": 1, "angle_deg": 360, "pulse_width_ms": 10}, "nmea": {"locked": 1, "polarity": "ACTIVE_LOW", "ignore_valid_char": 1, "baud_rate": "BAUD_9600", "leap_seconds": 27, "diagnostics": {"decoding": {"utc_decoded_count": 3495, "date_decoded_count": 3494, "not_valid_count": 1, "last_read_message": "GPRMC,032814.00,A,4013.03552,N,08308.98462,W,0.0,168.7,280520,,,d*65"}, "io_checks": {"start_char_count": 3556, "char_count": 251061, "bit_count": 742506, "bit_count_unfilterd": 742505}}}}}

TaylorDale, 2020-06-29:
To add a little more info here - We're experiencing issues with TIME_FROM_SYNC_PULSE_IN and I wonder if it is generally related. Using the latest ROS driver (v1.13) as soon as the sync pulse in starts, the ouster will lock up for about 10 seconds, and then reboot with internal time from OSC mode re-enabled (we have double checked that the settings config is rewritten when we make the changes).

When we revert the driver to the one we were using prior to the latest update, this issue does not occur and the Time sync from pulse in works very well.

dgarbuzov, 2020-12-07:
This is odd -- `TIME_FROM_SYNC_PULSE_IN` shouldn't really be interacting with the driver code, which just passes hardware timestamps through to the user. Is this still happening with the latest sample code?

TaylorDale, 2020-12-08:
I'll have to get back in the coming weeks, we are about to start up on this project again. Cheers.

Krishtof-Korda, 2021-04-22:
@TaylorDale bump on this. Were you able to resolve this with the v2.0 driver and firmware?

closed 2021-04-30 by dgarbuzov