ouster-lidar/ouster-sdk #167: ROS Noetic Ninjemys Support

NathanDodson, 2020-05-28:
Hi there! With the release of the newest version of ROS 1, I was wondering if this repo will be made compatible with Noetic? My team would like to try to update to Noetic, but I see that this package is listed as a blocked release at this link: 'http://repositories.ros.org/status_page/blocked_releases_noetic.html?s=6&r=1'. Please let me know if there is a plan for this or an equivalent repo has already been ported to Noetic.

Thanks,
Nathan D.

dllu, 2020-05-29:
the next release will support ros noetic along with

* support for gen 2 sensors (OS0, OS1-128, OS2, etc)
* new visualizer running on raw OpenGL/GLFW instead of VTK, supporting macOS, Windows, and Linux

the github may be updated with that soon

haritha-m, 2020-09-09:
Hi Ouster team, 

Wondering if there is any update on making the Ouster ROS node work with Noetic?  

Thank you!

ThomasEmter, 2020-09-14:
Hi Ouster team,

any news on the support of the gen 2 sensors?

Thank you.

dgarbuzov, 2020-12-07:
Latest update (including noetic support) was merged last week. Appreciate your patience

closed 2020-12-07 by dgarbuzov