ouster-lidar/ouster-sdk #171: Bug report with 1.14-beta10

TixiaoShan, 2020-06-19:
I am using the 1.14-beta10 with OS1-128, I have observed the following two problems:

- "TIME_FROM_PTP_1588" timestamp_mode is not working. OS1-128 still reports timestamp from the internal OSC. The same configuration works with OS1-64-gen1 and v1.13.
- There is a "ring" in the center of the point cloud. The diameter is roughly 0.03m.

[image]

TixiaoShan, 2020-07-02:
"TIME_FROM_PTP_1588" mode seems to be working now. But it takes at least 2 minutes to sync.

cynosure4sure, 2020-09-17:
I am  also having the same issue with a OS0-128 lidar. The lidar is on firmware 1.14.0-beta.12. The observation is the same that a ring appears in the 2-3cm range. One thing I noticed is that the pointcloud in ROS has no NaN or Zero points in it which is a bit odd that there are no invalid points in the cloud. Also the ring contains returns from multiple scanlines and not just a single one. Is it possible that invalid points are published on this ring?

[image]

[image]

dgarbuzov, 2020-12-07:
the ring issue should be fixed with #193. Please re-open otherwise

closed 2020-12-07 by dgarbuzov