ouster-lidar/ouster-sdk #175: Configure multipurpose_io pin as output

apabu, 2020-07-13:
Hi, I'm new to ROS and it will be great if someone could help me out with this. I am trying to configure the multipurpose_io pin as an output which will give me a signal at 10hz (a pulse every time the lidar completes a full rotation). I went through the software user guide and figured out that using the below will activate the pin and output a pulse that I require. 

set_config_param multipurpose_io_mode OUTPUT_FROM_ENCODER_ANGLE
set_config_param sync_pulse_out_angle 360
reinitialize

When I try to check the output signal from the multiputpose_io pin, I get a flat line. I'm not sure what I am doing wrong. 
It'll be great if someone can help me out. TIA!

yulan0215, 2021-03-09:
> Hi, I'm new to ROS and it will be great if someone could help me out with this. I am trying to configure the multipurpose_io pin as an output which will give me a signal at 10hz (a pulse every time [...]

I encountered the same problem did you solve it? I used the multipurpose_io for NMEA input but I also want to output the PPS signal, do you think it is feasible? Thanks and I am looking forward to you reply!

apabu, 2021-03-09:
If you look at the end of the hardware user guide of OS1 (https://data.ouster.io/downloads/hardware-user-guide-v1.13.0.pdf), it provides a hint about amplifying the signal from the multipurpose_io. When I measured the signal, I figured out that the output is less than 30mV. I needed a signal of amplitude 3.3-5V, so I used an op-amp to amplify the signal. I used this signal for time synchronization with cameras. I'm not sure if you can use the pin for both NMEA input and PPS output at the same time. 
[image]

yulan0215, 2021-03-09:
> If you look at the end of the hardware user guide of OS1 (https://data.ouster.io/downloads/hardware-user-guide-v1.13.0.pdf), it provides a hint about amplifying the signal from the multipurpose_io. [...]

Thanks anyway. I guess it is not feasible because there is just one mode for the multipurpose, if I select input it could not output but I do not know.

Krishtof-Korda, 2021-04-22:
> If you look at the end of the hardware user guide of OS1 (https://data.ouster.io/downloads/hardware-user-guide-v1.13.0.pdf), it provides a hint about amplifying the signal from the multipurpose_io. [...]

You are correct on the need for amplification. This is something we are improving in our next revision of the sensor. 
You cannot use the M_IO pin for both NMEA input and PPS output.

closed 2021-04-22 by Krishtof-Korda