ouster-lidar/ouster-sdk #176: Question about ptp and step change of time

msadowski, 2020-07-16:
Hi!

I'm running into a bit of an issue with ptp. I've set up my single board PC to get the time from GPS NMEA sentences + PPS and I wanted to use this time to drive the clock of the ptp (and thus Ouster).

Since the system does not have an RTC every time it boots the time is set to January 2018, only after GPS gets some satellites in view it manages to correctly set the system clock, however the new time is not reflected on the Ouster side. I've tried setting the step_threshold in ptp to some value expecting the time to jump, however it is not the case. I've also tried restarting ptp4l while the sensor is on but the clock would not change. When debugging with rostopic delay I see the following output:

```
average delay: 77768078.034                                                              
        min: 77768073.859s max: 77768092.056s std dev: 2.41305s window: 19295                                                                         
average delay: 77768078.012                                                                                                                                                           
        min: 77768073.821s max: 77768092.056s std dev: 2.42561s window: 19396                                  
average delay: 77768077.990                                                                                                                      
        min: 77768073.783s max: 77768092.056s std dev: 2.43824s window: 19498                                                                                                
average delay: 77768077.968                                                                                                                                                        
        min: 77768073.746s max: 77768092.056s std dev: 2.45068s window: 19599                                                                                                                
average delay: 77768077.946                                                                                                                                                   
        min: 77768073.708s max: 77768092.056s std dev: 2.46321s window: 19701                                                                                                                 
average delay: 77768077.924                                                                                                                          
        min: 77768073.670s max: 77768092.056s std dev: 2.47567s window: 19803                                                                                                                 
average delay: 77768077.902                                                                                                                                                                   
        min: 77768073.633s max: 77768092.056s std dev: 2.48797s window: 19904                                                                                                             
average delay: 77768077.881                                                                                                                                             
        min: 77768073.596s max: 77768092.056s std dev: 2.50022s window: 20005                                                                                                                
average delay: 77768077.859                                                                                                                             
        min: 77768073.558s max: 77768092.056s std dev: 2.51254s window: 20107                                                                                                                 
average delay: 77768077.837                                                                                                                          
        min: 77768073.521s max: 77768092.056s std dev: 2.52469s window: 20208                                                                                                                 
average delay: 77768077.815                                                                                                                                                                   
        min: 77768073.482s max: 77768092.056s std dev: 2.53704s window: 20311
...
```

From this you can see that the clock is very slowly converging, however it will never get there by just frequency modifications.

Only if the system booted and I powercycle the Ouster the time is set correctly.

Did anyone come across this? I'd be grateful for any hints or workarounds!

dsb6063, 2020-07-16:
If your setup is similar to others that had the time issues, the board needs battery support to reduce the GPS-start to lock during boot. Basically, the GPS would be on all the time or have access to power. Otherwise, there are similar inexpensive boards that handle the RTC and hold the value during power off. This is assuming the GPS unit is only used for timing and not an expensive GPS unit for precise positioning.

msadowski, 2020-07-17:
Many thanks for the reply! I was actually planning on using the GPS for both timing and positioning. Is there no other way to reset the ptp clock on Ouster other than powercycle?

If there is no way then I'll go with an RTC, which sounds like a good practice anyway.

adityarath302, 2020-07-23:
@msadowski could you elaborate how did you set up the PTP on a Linux machine to syn with your gps and ouster unit.

msadowski, 2020-07-23:
I've followed [Software Guide](https://data.ouster.io/downloads/software-user-guide-v1.13.0.pdf) for ptp+phc2sys setup. For GPS + PPS synchronization I roughly followed this page: https://www.satsignal.eu/ntp/Raspberry-Pi-NTP.html.

dsb6063, 2020-07-23:
I did also.

Krishtof-Korda, 2021-01-07:
Cześć @msadowski,

The ouster sensor only "jumps" time once after power-up. This is by design to prevent time jumping due to a gross malfunction of the grandmaster. 

It is recommended that you only power on the ouster sensor after the gm is announcing the correct time.

Another option is to try a beta feature that is on firmware v2.0. You can set the ptp profile of the ouster sensor to "jump" time whenever the offset is greater than 1 sec. Please reference the article, [How to enable PTP Profiles on Ouster Sensor](https://ouster.atlassian.net/servicedesk/customer/kb/view/1165066356)

Cheers!

asherikov, 2021-01-07:
@Krishtof-Korda , I think it would be convenient to perfrom jump by request, although it is probably not trivial to implement. Does  'automotive-slave' perform jumps if offset > 1sec?
A slightly unrelated question: I observe rather significant (but < 50ms) occasional divergence between the master and the lidar, is there something I can do to prevent this?

asherikov, 2021-01-07:
It also looks like PTP profile cannot be set with `set_config_param` as other parameters, which is rather inconvenient.

Krishtof-Korda, 2021-01-08:
@asherikov , Only the `"default-relaxed"` and `"gptp-relaxed"` profiles perform jumps if offset > 1 sec. 

Regarding the < 50 ms occasional divergence; what are you comparing to arrive at that number? Check out our tool for accurately measuring the offset in this comment,  https://github.com/ouster-lidar/ouster_example/issues/78#issuecomment-755828437

Regarding PTP profile settings, we plan to have parity between HTTP API and TCP API in the future, but for now, it can only be configured via the HTTP API. Thanks.

asherikov, 2021-01-08:
Thanks for reply. I am checking `current_data_set/offset_from_master` returned by `api/v1/system/time/ptp/`, I am polling API once in two seconds, it usually takes 10-20 second to converge back to <5ms, but I haven't tried PTP profiles yet.

Krishtof-Korda, 2021-01-09:
@asherikov
Roger that. Would you mind posting the output of `api/v1/system/time/ptp/`?
Keep in mind the value for `offset_from_master` is reported in nanoseconds. 
It should always remain < 1 ms. Usually on the order of 10s of microseconds.

asherikov, 2021-01-11:
Usually the offset is within [-0.5, 0.5] ms. I'll flash the latest firmware and if the problem persists I'll open a new github issue (let me know if you prefer something else).

closed 2021-04-30 by dgarbuzov