ouster-lidar/ouster-sdk #177: Ambient and relfectivity data in Ouster Lidar

ajay1606, 2020-07-17:
Hello all,
I recently started to work with OS 2 128 channel lidar, and range and better choice for LiDAR-based object perception.
But I have a few questions related to ambient and reflectivity data from the sensor. 

 How the ambient and reflectivity calculated ? is there any research publication that describes the method. 

Thanks in advance,
 
Regards,
Ajay

dllu, 2020-07-17:
As you may know, the lidar works by measuring the number of photons as a function of time (at the scale of nanoseconds). Once you have such a time series, the ambient, intensity, and range are related as shown in this diagram.

[image]

Currently, the reflectivity is calculated assuming it is proportional to intensity * range^2, which is an approximation that according to assuming the inverse square law for Lambertian objects in the far field.

ajay1606, 2020-07-17:
@dllu Thank you so much for your quick response. Understood the concept.

closed 2020-07-17 by ajay1606