ouster-lidar/ouster-sdk #178: GPS Sync with Ouster Lidar OS1-32

adityarath302, 2020-07-22:
Hello,
I have been working with Ouster 32 Channel Lidar for quite a sometime and I am stuck at a point where I need to sync a GPS Module with the Lidar. For a good SLAM, I brought a REACH M2 and now I have tried everything given in the software guide but still, I am not able to get the GPS to sync. I tried everything given on the Ouster Forum but still, the NMEA Locked variable is not turning to 1 as below. I think the issue is with the NMEA message received as this GPS supports GNRMC but ouster Supports GPRMC. Could anyone help me, please!!
PS:If PTP works where a Linux machine like raspberry pi is the master and GPS and Ouster act like a slave then please share the method with me.
Ref for GPS:https://forum.ouster.at/d/74-how-to-synchronize-gps-with-ouster-sensor
[image]

michielvlaminck, 2021-01-25:
I have the same problem. Were you able to solve it in the end?

yulan0215, 2021-03-08:
> I have the same problem. Were you able to solve it in the end?

I had the same problem, did you solve it?

michielvlaminck, 2021-03-08:
Yes, I believe the following TCP command did the trick:
`set_config_param nmea_ignore_valid_char 1`

Also make sure that the baud_rate of your GNSS matches the one of the Ouster scanner (and try both 9600 and 115200).

yulan0215, 2021-03-08:
> Yes, I believe the following TCP command did the trick:
> `set_config_param nmea_ignore_valid_char 1`
> 
> Also make sure that the baud_rate of your GNSS matches the one of the Ouster scanner (and try both 9600 and 115200).

Thank you for your quick reply!
Do you mean input this  TCP command into the shell?

michielvlaminck, 2021-03-08:
Yes, exactly.

You're welcome.

yulan0215, 2021-03-15:
> Yes, exactly.
> 
> You're welcome.

Thx! I am sorry that is there any method to valid that the timestamp of Ouster LiDAR come from GPRMC? I will appreciate if you know it and tell me! Thx!

michielvlaminck, 2021-03-15:
You can use Ouster Studio and check the timestamp there?

yulan0215, 2021-03-15:
I used Ubuntu system.
Actually I would like to convert the format of the timestamp from Unix format into date to check if it is the time same as GMT. I used the Ouster driver for ROS system and I got the date by accident but when I would like to repeat it again, it did not work. I guess there was a command for the output of date. Like the picture I showed below:
[image]

Do you know any command related to this? Thanks!

closed 2021-04-30 by dgarbuzov

AchmadFathoni, 2024-02-17:
Using firmware 2.3.5 on Rev-D lidar with `nmea_ignore_valid_char` set 1 still won't make Ouster decode GNRMC message. Ouster only able to lock and read the message. Here the attached json [time.json](https://github.com/ouster-lidar/ouster_example/files/14318819/time.json)

I know it is wrong place to complain about closed-source firmware but since the forum won't let me make new account because CAPTCHA, I have no other option.