ouster-lidar/ouster-sdk #181: [WARN]: Unknown sensor firmware version; output may not be reliable

SubMishMar, 2020-08-03:
Hi,
I am trying to interface an Ouster (OS1) 128 channel lidar with ROS. I can ping the sensor and there is communication but I am not able to get any data from the sensor. 
Below you can see what messages I get on launching the os1.launch file. 
[image]

When I echo the required topics I see no messages. Any idea what's wrong?

Thanks,
Subodh.

adityarath302, 2020-08-07:
Try Changing the firmware on the device. Preferably v1.13.

dllu, 2020-08-07:
**Please do not install firmware v1.13 on your OS1-128.** The GitHub code hasn't been updated to support the OS1-128 and other gen 2 sensors yet. Contact Ouster support for a link to the prerelease code that works with gen 2 devices.

SubMishMar, 2020-08-07:
@dllu yes they have got back to me and I understand that the old stack wont work with gen2. So yes, I am not installing 1.13. Thanks for your response :)

closed 2020-12-07 by dgarbuzov