ouster-lidar/ouster-sdk #184: Timestamp of each point in the point cloud

SubMishMar, 2020-09-22:
Hi,

Is it possible to obtain the time-stamp of each point in the /os_cloud_node/points topic? Although there is a `t' field for each point, it outputs zero when I try to print it (using std::cout or ROS_INFO_STREAM).

To the best of my understanding every column a 360 scan has 4 different timestamps which comes from 4 different azimuth angles within a single column of observations.

Please let me know how I can access the time stamps associated with every point.

Regards,
Subodh.

SubMishMar, 2020-09-22:
I realized now that the time field of a point equals `ts-scan_ts`. I will be glad to know why is it so? Why is it not just `ts`?

dllu, 2020-09-24:
It was set to be ts - scan_ts because 

* absolute timestamps in IEEE 1588 PTP time sync mode or GPRMC/PPS time sync mode are UNIX timestamps that may overflow the 32 bit integer
* it maintains compatibility with some 3rd party packages such as LOAM-type slam algorithms that "dewarp" or "deskew" the point cloud based on the time since the start of the scan

SubMishMar, 2020-09-24:
Thanks! @dllu

closed 2020-09-24 by SubMishMar

BritskNguyen, 2020-10-01:
> I realized now that the time field of a point equals `ts-scan_ts`. I will be glad to know why is it so? Why is it not just `ts`?

Hi,

May I ask where found this definition, and what **ts** and **scan_ts** actually means with respect to the original defintions (**header.stamp** and **p.t**)?

Thanks

gogojjh, 2022-01-28:
> I find this sentence in https://github.com/ouster-lidar/ouster_example/blob/13ea8e8b8a4951fb630dbc9108666995c8443bf6/ouster_ros/src/ros.cpp?_pjax=%23js-repo-pjax-container%2C%20div%5Bitemtype%3D%22http%3A%2F%2Fschema.org%2FSoftwareSourceCode%22%5D%20main%2C%20%5Bdata-pjax-container%5D#L126

Alex-Beh, 2022-03-03:
@gogojjh @brytsknguyen Would you all mind to elaborate the meaning behind `ls.header(v).timestamp` and `scan_ts`? I suspect `scan_ts` is the timestamp of the first firing in current cycle.

themightyoarfish, 2023-03-29:
It looks like `const auto ts = (ls.header(v).timestamp - scan_ts).count(); ` is actually the relative time difference, i.e. the first point would have ts = 0.