ouster-lidar/ouster-sdk #187: OS1-64 restarts or pauses when os1.launch called

GrantZheng86, 2020-10-01:
I noticed that whenever I call the "os1.launch", the flashing red light in the LIDAR seemed to be turned off. It will turn back on after about 10 seconds. I am running ubuntu 18, with the most recent "ouster_example" files from this repo. There is a computer in my lab that is running ubuntu 16 with a much older version of the files from "ouster_example" (should be around 9 months to 1 year ago), and there was no restarting or pausing for the lidar. This restart or delay is actually causing problems in one of my big launch files, and it's throwing out errors `[FATAL] [1601500521.299787039]: F0930 15:15:21.000000 18745 ordered_multi_queue.cc:112] Check failed: last_dispatched_time_ <= next_data->GetTime() (621355973168931677 vs. 621355968400000656) Non-sorted data added to queue: '(2, /points2)'` I am wondering what function is likely to cause this restart or delay? From the terminal output, it looks like rewriting of the metadata is likely to cause the problem, but I am not sure. Does anyone have any ideas about this?
 
Thanks for helping out!

Also, the code below is from the old os1.launch, which has no problem in resetting the lidar.

`<launch>

  <arg name="respawn" default="false" />
    
  <arg name="scan_dur_ns" default="100000000" doc="nanoseconds to batch lidar packets before publishing a cloud"/>
  <arg name="os1_hostname" default="os1-991837000025" doc="hostname or IP in dotted decimal form of the sensor"/>
  <arg name="os1_udp_dest" default="192.168.1.2" doc="hostname or IP where the sensor will send data packets"/>
  <arg name="os1_lidar_port" default="7502" doc="port to which the sensor should send lidar data"/>
  <arg name="os1_imu_port" default="7503" doc="port to which the sensor should send imu data"/>
  <arg name="replay" default="false" doc="when true, the node will listen on ~/lidar_packets and ~/imu_packets for data instead of attempting to connect to a sensor"/>

  <group ns="/os1_node">
    <param name="scan_dur_ns" value="$(arg scan_dur_ns)"/>
    <param name="os1_hostname" value="$(arg os1_hostname)"/>
    <param name="os1_udp_dest" value="$(arg os1_udp_dest)"/>
    <param name="os1_lidar_port" value="$(arg os1_lidar_port)"/>
    <param name="os1_imu_port" value="$(arg os1_imu_port)"/>
    <param name="replay" value="$(arg replay)"/>
  </group>

  <node unless="$(arg respawn)" pkg="ouster_ros" type="os1_node" name="os1_node" output="screen" required="true" />
  <node if="$(arg respawn)" pkg="ouster_ros" type="os1_node" name="os1_node" output="screen" respawn="true" />

</launch>`

dgarbuzov, 2020-12-07:
The provided launch file and ROS code will unconditionally re-initialize the sensor on startup, after updating the configuration parameters. Iirc, this has always been the case, so I can't explain why it might have worked with an old version.

If you want to avoid any gaps in the udp data stream on startup, you can modify the os_cloud_node to use the second overload of `init_client`:
https://github.com/ouster-lidar/ouster_example/blob/9352730127c3e2bc8ce1e778be4ae8b0a3b40f3c/ouster_client/include/ouster/client.h#L38

This will avoid reinitializing the sensor, but can't be used to update the configuration -- you'll have to e.g. set the lidar_mode separately.

GrantZheng86, 2020-12-12:
Thanks for the advice, I'll Give it a try!

closed 2020-12-12 by GrantZheng86