ouster-lidar/ouster-sdk #191: os1-gen2-32 ubuntu melodic got "Unexpected udp packet length: 6464"

Eric97, 2020-11-26:
Hello,

I am testing a new OS1-gen2-32 with firmware v1.14.0 under Ubuntu 18.04 melodic. The lidar and PC are connected through a local router and PC can ping through the lidar. When launching 
` roslaunch ouster_ros os1.launch os1_hostname:=192.168.1.101 os1_udp_dest:=192.168.1.100 lidar_mode:=512x10` , the "Unexpected udp packet length: 6464" error kept coming out. 
No `/os1_cloud_node/points` were received, but imu topic can be received. What might be the reasons here? Thanks.

dllu, 2020-11-27:
The GitHub code does not currently support Gen 2 sensors. It will be updated soon to support Gen 2 sensors. Until then, you will need to download the tarball from here: https://ouster.com/beta-program

dgarbuzov, 2020-12-07:
Gen2 / fw 2.0 firmware as added in #193. Please re-open if you're still having issues

closed 2020-12-07 by dgarbuzov