ouster-lidar/ouster-sdk #192: Unsynchronization of Ouster and PC (both are connected by a router)

ArlenCHEN, 2020-12-02:
Hi all,

I am having a timestamp issue. My Ouster lidar is not directly connected to my PC, instead both are connected to a router, which assigns IPs to the two ends. With this setting, it seems the time on Ouster is counted from turning-on, although I already set the timestamp mode as `TIME_FROM_PTP_1588`. I want the time on Ouster can be synced with my PC, where I need to use ROS. 

Below gives the information from `get_time_info`:
`{"timestamp": {"time": 735.84418612, "mode": "TIME_FROM_PTP_1588", "time_options": {"ptp_1588": 735, "sync_pulse_in": 0, "internal_osc": 725}}, "sync_pulse_in": {"locked": 0, "polarity": "ACTIVE_HIGH", "diagnostics": {"last_period_nsec": 0, "count": 0, "count_unfiltered": 1}}, "multipurpose_io": {"mode": "OFF", "sync_pulse_out": {"polarity": "ACTIVE_HIGH", "frequency_hz": 1, "angle_deg": 360, "pulse_width_ms": 10}, "nmea": {"locked": 0, "polarity": "ACTIVE_HIGH", "ignore_valid_char": 0, "baud_rate": "BAUD_9600", "leap_seconds": 0, "diagnostics": {"decoding": {"utc_decoded_count": 0, "date_decoded_count": 0, "not_valid_count": 0, "last_read_message": ""}, "io_checks": {"start_char_count": 0, "char_count": 0, "bit_count": 1, "bit_count_unfilterd": 0}}}}}`

Anyone can give some clue to solve this issue? Thanks!

kylemanna, 2020-12-05:
Are you running a PTP grandmaster clock on a device on the network?  That's necessary for the sensor to get the time.

See the [User Manual](https://data.ouster.io/downloads/software-user-manual/software-user-manual-v2p0.pdf) for details. In particular
*  `GET /api/v1/time` to view the PTP status
*  "PTP Quickstart Guide"

ArlenCHEN, 2020-12-16:
@kylemanna Thanks! 
Yes, the PTP grandmaster clock is running on the PC, but the problem is still there if the Ouster and the PC are communicated with a router. Not sure if there is any solution to this issue with a router, but I solved the problem by just directly connecting the Ouster to the PC. It works fine to me now.

closed 2020-12-16 by ArlenCHEN