ouster-lidar/ouster-sdk #194: ROS Service to configure the Lidar

ckurtz22, 2020-12-08:
Would be really great if we had a ROS service that allowed us to configure the Lidar, or at the very least do stuff such as put the Lidar in/out of standby mode.

msadowski, 2020-12-09:
I'm also very interested about being able to put the lidar in 'standby' or 'pause' modes so that no lidar information is published. Is anything like that available already?

dgarbuzov, 2020-12-09:
to clarify - you'd also want configuration (including standby mode) available specifically as a [ROS service](http://wiki.ros.org/Services)?

ckurtz22, 2020-12-09:
> I'm also very interested about being able to put the lidar in 'standby' or 'pause' modes so that no lidar information is published. Is anything like that available already?

Yeah the newest 2.0 firmware added a configuration that allows you to put the lidar in standby where it stops the motor and laser and only draws 5W of power. I'm not sure whether it stops sending packets, haven't checked.

kairenw, 2021-04-27:
Hi @ckurtz22, 

The newest master has [an early version of an API ](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/client.h#L124 ) covering the full sensor configuration. It may change, and it isn't available as a ROS service, but thought it might interest you. Let us know if the functionality is useful!

ckurtz22, 2021-04-30:
> Hi @ckurtz22,
>
> The newest master has [an early version of an API ](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/include/ouster/client.h#L124) covering the full sensor [...]

Oh, that's sweet! Thanks for letting me know. I may make a PR adding a ROS service at some point but haven't had much time to work on my project using this lately.

RFRIEDM-Trimble, 2021-06-08:
I'd be happy to 

> I'm also very interested about being able to put the lidar in 'standby' or 'pause' modes so that no lidar information is published. Is anything like that available already?

I was also looking into this. In ROS2, there explicit steps for set up and tear down of the connection, in which TCP commands could be issued to turn the sensor in `NORMAL` when the driver node starts running and `STANDBY` when the driver stops. I don't see any reason the sensor should be spinning if there isn't something listening. 

I am also interested in adding the configuration as a ROS2 service. It does already support `GetMetadata`, but no setting of parameters from a ROS2 service already running. @ckurtz22 Are you interested in ROS1 or ROS2?

jeremyroy, 2021-12-14:
I am also interested in this feature, for ROS1

skarful, 2021-12-15:
Hi everyone, I was looking at something similar for my application. I used the API mentioned by @kairenw and created a ROS service for sending config params to the lidar. I've sent in a pull request (https://github.com/ouster-lidar/ouster_example/pull/331). Until it gets reviewed, you can use the ros_config_service branch forked on my github to accomplish this (https://github.com/skarful/ouster_example/tree/ros_config_service). I've tested this with the OS1 and the OS2 lidars. Hope it helps :)

Samahu, 2022-08-25:
@ckurtz22 and everyone else who might interested, the feature requested has been implemented and is part of the last release. As I responded to @skarful on the PR #331 the functionality is the same with few differences.

The new service which is named `/ouster/set_config` takes a path to a configuration file rather than a string, another thing worth noting is that while the service operational it is still experimental since there are certain scenarios that does not supported. 

Please refer to the PR #420 for additional details.

Since the requested feature is now part of the SDK i will close this issue.

closed 2022-08-25 by Samahu