ouster-lidar/ouster-sdk #196: Saving as .pcap or .las using a script

NiharMasurkar, 2020-12-10:
Hi, I have an Ouster OSO-32(Gen-2) LIDAR which is connected to a system. 
I want to write a script to save the LIDAR packets in the form or a pcap file so that I can store them in a remote location for further use. 

I can use the Ouster Studio to save the files as a pcap file, but I was looking for a terminal line command such as TCPDUMP or Wireshark to save these pcap files. 

Also, I was hoping to save these files as a .las for a mapping project that needs a las file format for processing the data. 
Does anyone have a solution for this?

sam-rke, 2021-01-15:
Hi @NiharMasurkar , have you been successful in your scrip part?
if so kindly enlighten us.

dgarbuzov, 2021-01-16:
tcpdump should work for recording pcaps -- are you looking for additional information, step-by-step instructions, or something else?

sam-rke, 2021-01-18:
Hi @dgarbuzov !
thankyou for "tcpdump" suggestion.
I think it is from tcp api list.
let me check first, if I face any issue I'll get back to you again.
Thankyou for your support.

sam-rke, 2021-01-18:
> Hi @dgarbuzov !
> thankyou for "tcpdump" suggestion.
> I think it is from tcp api list.
> let me check first, if I face any issue I'll get back to you again.
> Meanwhile if you have any step by step guide link for capturing pcap with code only, kindly share that. As for now i only know that pcap can be generated with ouster studio only.
> Thankyou for your support.

Just to clear my doubt "tcpdump is a command line utility that allows you to capture and analyze network traffic going through your system". But @dgarbuzov  I am looking for "How to capture pcap with ouster driver code or how to convert .bag to .pacap with code or library"

dgarbuzov, 2021-01-18:
Driver code isn't necessary to record pcaps -- they just contain packets sent over the network. But if you want to replay data in Ouster Studio, you'll have to record them with Ouster Studio. For help with Ouster Studio, please contact Ouster support directly.

Re: converting bags to pcaps -- we don't currently provide any code to do that in this repository.

closed 2021-01-18 by dgarbuzov