ouster-lidar/ouster-sdk #197: How to persist the changes for PTP synchronization with LinuxPTP Grandmaster and what is the use chrony?

afridi26, 2020-12-14:
Hey everyone,

I follow the [official](https://data.ouster.io/downloads/software-user-manual/software-user-manual-v2p0.pdf#52) documentation to setup PTP for the Os1-64 (GEN 1). I followed the documentation till the end of section 15.6.3 (Configuringphc2systoSynchronizetheSystemTimetothePTPClock) and everything is working fine.

However,  when the system is rebooted or shutdown and started again, I get the following output for the commands (_I did debugging to find the root cause of the problem_):

[image]

similarly,
[image]
and 
[image]

However, When I reload the daemon, it starts working again:
[image]

So there are now two questions,
1) How to persist the changes, so when I restart the system, the ptp4l should be working such I should not run the 
sudo systemcl daemon-reload on every restart.

2) What is chrony? Which is mentioned in official documentation? Should I need to configure it also because Iskipped this section (15.6.4Configuring Chrony to Set System Clock Using PTP) the ptp4l.

Any help would be appreciated :)

Thanks.

afridi26, 2020-12-14:
Fixed typo

afridi26, 2020-12-23:
Nevertheless, I solved the problem by enabling the service as follows (I forget to enable the ph2sys service):

`sudo systemctl enable phc2sys.service`

and the run

`sudo systemctl daemon-reload`

Thanks

afridi26, 2021-01-26:
Hey,

the problem still persist? Can anyone guide how this can be fixed?

Regards,
Aftab

dgarbuzov, 2021-04-30:
That warning about `systemctl daemon-reload`being needed only happens when the service files are modified. If that always happens on boot, something may be mis-configured on your system and overwriting those files. Could double-check the mtime to make sure.

closed 2021-04-30 by dgarbuzov