ouster-lidar/ouster-sdk #198: Ring Index in ROS driver

AndreasInSchwaben, 2020-12-16:
Hi,

if the ROS driver's topic would include a ring index, more advanced post-processing would be possible. The ring-index allows to identify each Lidar point to the transmitted beam. 
See e.g.: https://answers.ros.org/question/132811/accessing-layers-in-velodyne-point-cloud/  

would be great to have this feature available!
Thanks a lot!
Andreas

dgarbuzov, 2020-12-16:
If I understand the question, it's there already:
https://github.com/ouster-lidar/ouster_example/blob/5c00aa97ee8244072ed9bf14870facf57d4aab98/ouster_ros/include/ouster_ros/point.h#L23
I'm able to select `ring` as a channel in rviz and visualize it, etc.

AndreasInSchwaben, 2020-12-17:
Dear all,

agreed!
The feature of the ring-number already exists in rosbags and RVIZ is able to display or visualize them.

However, in all the documentation  available to us, we have not found any hint on that feature.
Would you mind to update the documentation with the ring-numbers, please?

Thanks a lot & Merry Christmas
Andreas

https://github.com/ouster-lidar/ouster_example/blob/5c00aa97ee8244072ed9bf14870facf57d4aab98/ouster_ros/include/ouster_ros/point.h#L23<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_ouster-2Dlidar_ouster-5Fexample_blob_5c00aa97ee8244072ed9bf14870facf57d4aab98_ouster-5Fros_include_ouster-5Fros_point.h-23L23&d=DwMCaQ&c=fP4tf--1dS0biCFlB0saz0I0kjO5v7-GLPtvShAo4cc&r=dSAqM5GQyTlYsmQj3qRmud4QxHm7vVxi9N5Jw0dC-98&m=iccH6KfShXoq8ecccimaRI-EUewckpzBcMD6dkRNyIY&s=TU5lPjFVt0SY6TwPmLspBds6E7NofgxyKPxWZkhLjAQ&e=>

AndreasInSchwaben, 2021-01-08:
Dear Ouster,

we are still waiting for an API documentation, how the Ring index may be read from Lidar data, your ROS driver or topic.

Best Regards
Andreas Schwager

Closed #198<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_ouster-2Dlidar_ouster-5Fexample_issues_198&d=DwMCaQ&c=fP4tf--1dS0biCFlB0saz0I0kjO5v7-GLPtvShAo4cc&r=dSAqM5GQyTlYsmQj3qRmud4QxHm7vVxi9N5Jw0dC-98&m=BnWksPr-8QJj93nfwDHeSXIKvzPfrXAK6XYwJFAuzZs&s=bbIn6VrSNP_H9wdZUq66wb97G5CShiaHgnsD3D2xvuo&e=>.

dgarbuzov, 2021-01-08:
We'll be updating and publishing API docs for the sample code as our schedule allows. Until then, if I'm happy to answer any specific questions you might have.

AndreasInSchwaben, 2021-01-12:
Dear Ouster Colleagues,

we are searching for an API definition how data are sorted in the ROS topic sent by your ROS-driver.
We are reading the ROS topic from Matlab where we do not have the support of e.g. RVIZ plugin.

Could you please send us any documentation how to read and interpret the ROS-topics send by Ouster ROS driver.

Thank you very much
Andreas

dgarbuzov, 2021-01-16:
Unfortunately we don't currently provide support for matlab in this repo. I'll reopen this issue in case other users that have some more experience reading ROS topics in matlab can share their solutions.

If I understand correctly, you're able to receive point clouds, but are having trouble getting the actual field data. Have you tried the [readField() function](https://www.mathworks.com/help/ros/ref/readfield.html)? It looks like it should allow you to read the additional fields defined for [our point type](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_ros/include/ouster_ros/point.h) like "ring". [This function to enumerate all available fields in a point cloud](https://www.mathworks.com/help/ros/ref/readallfieldnames.html) might also be useful.

closed 2021-12-28 by dgarbuzov