ouster-lidar/ouster-sdk #200: About Ouster Lidar

sam-rke, 2020-12-26:
Hi there,
I am new with ouster lidar.
i need to know if this code can be used with Ouster OS0 as well.
Second this is do we have some kind of simulator for Ouster and Nvidia jetson TX2.
So that I can test written code untill i receive the device.

sam-rke, 2020-12-29:
Hi @dllu (Daniel),
Thanks for your response.
I am just trying to understand this code.
I am not sure if i have to modify it or it can be directly integrated into my project.
Can you suggest something in this. I would be very grateful
Thanks

dllu, 2020-12-29:
The code is open source. Feel free to modify it or integrate it into your project however you like.

sam-rke, 2021-01-07:
hi @dllu thankyou for your response.
Lidar will be reaching soon but i am still having some doubt.
there is an executable in ouster example part within which i have to give lidar address and host address.
now my question is if i need to save data to local storage like sd card of jetson why i need to have a host address secondly will lidar save data automatically to local storage.
can you please guide something in it?

dgarbuzov, 2021-01-07:
The code here currently only supports replaying data using ROS. See the README for details on how to replay sample data provided as ROS bags.

closed 2021-01-07 by dgarbuzov