ouster-lidar/ouster-sdk #201: Clairfy Fig 3.1 in software user manual v2.0

versatran01, 2020-12-29:
I have problem understanding the equations that compute xyz in page 12 of software user manual 2.0
[image]

[image]

[image]

So my question is, what is `r`? Is it the distance to the beam origin (the rotating block)?
If so, then why is `n` involved in the computation of `z`?

dllu, 2021-01-04:
The definition of `r` is literally right above the equations that you screenshotted.

[image]

versatran01, 2021-01-04:
I saw that but is it range to the beam origin or to the center of the sensor?

versatran01, 2021-01-05:
@dgarbuzov any comments?

dgarbuzov, 2021-01-07:
The reported range `r` is the full distance to the center of the sensor, including `n` (the distance between the lidar origin and beam origin). In the equations for `x` and `y`, `r - n` is the measured range from the beam origin, so the first term gives you the coordinate in the frame of the beam origin and and the second term accounts for the offset to the lidar origin.

Hope that helps. I agree the label for `r` in the diagram is misleading -- sorry about that!

closed 2021-01-07 by versatran01