ouster-lidar/ouster-sdk #206: Homologation of ouster client for ROS / ROS 2 support

SteveMacenski, 2021-01-07:
How can we combine the changes that we've made in ROS2 to enable different time sources into the ouster_client so that we can make migration between firmware version easier

SteveMacenski, 2021-01-07:
After, it would be good to have `ouster_client` in its own repository we can release as binaries so that both the ROS1 and ROS2 drivers can share them without copy-paste action.

dgarbuzov, 2021-01-16:
You mean making the timestamp mode configurable? If I understand the changes in the ROS2 repo correctly, we've already added that functionality here:

https://github.com/ouster-lidar/ouster_example/blob/5c00aa97ee8244072ed9bf14870facf57d4aab98/ouster_client/include/ouster/client.h#L51

The only part that's missing is the "fake" mode that sets timestamps based on the host clock. Happy to add that here if it's necessary.

Re: splitting out ouster_client: is that really necessary if it's possible to build *just* the client binaries from this repo?

SteveMacenski, 2021-01-19:
> You mean making the timestamp mode configurable? If I understand the changes in the ROS2 repo correctly, we've already added that functionality here:

Yes, but I think we added a couple of new options including using ROS time. If you've added the different modes, awesome! Just need to have all the options [Tom added](https://github.com/ros-drivers/ros2_ouster_drivers#timestamp-modes). 

> Re: splitting out ouster_client: is that really necessary if it's possible to build just the client binaries from this repo?

For versioning, probably it would be good software practice? For making the client code releasable for both the ROS 1 and ROS 2 drivers, yes. I can't have the ROS 2 drivers build on the build farm without it being released itself already on `apt` or as a separate repo I can myself release via the ROS build farm to then link against in the ROS 2 drivers - since we want to share it ideally.

RFRIEDM-Trimble, 2021-06-03:
Can you use cmake FetchContent or git submodules instead of copy/paste?

SteveMacenski, 2021-06-03:
No, they do not work for the ros build farm to release binaries