ouster-lidar/ouster-sdk #224: PTP divergence

asherikov, 2021-02-17:
Hello, I observe rather significant occasional divergence between the master and the lidar (`current_data_set/offset_from_master` returned by `api/v1/time/ptp`) when using PTP for synchronization, see the plot below for an example.

Configuration
- model: "OS-1-16-A0"
- firmware: "ousteros-image-prod-aries-v2.0.0+20201124065024"
- PTP profile: "default-relaxed"
- master is synced to GPS time

The issue is not new though and has been observed with older firmware versions as well. I would appreciate any help or suggestions.

[image]
Offset is in second (`offset_from_master` divided by 1e9).

@Krishtof-Korda this is the issue I mentioned in #176, please let me know if you need more info.

Krishtof-Korda, 2021-03-29:
@asherikov 
We suspect it's a known bug with some Intel NICs on the master or that the PTP master clock jumped time. We recommend the `default` profile instead of `default-relaxed`, that way you won't experience large divergences, instead the clock will gradually adjust frequency to any sudden master clock changes.

You should graph your master clock time changes, we suspect the jump originated there.
Cheers,
Krishtof

asherikov, 2021-03-30:
Thanks for reply,
1. NIC is Intel I210, can you provide links to the bug you have in mind?
2. The master is synced to GPS and we monitor its status, I doubt that jumps in the clock is the reason in this case, but I'll double check that too. 
3. When we used older version of the firmware, the only difference was that divergence was smoother, so I don't think that PTP profile is important.

Krishtof-Korda, 2021-04-21:
@asherikov 

First, I wanted to thank you for collecting the data in your previous graph. It is very helpful to have data like this to see the issue.

1. I don't have a link to a specific bug, but there are a few that are out there.
2. This is almost certainly a jump in the master or system time, our internal clock only does gradual frequency adjustments in the `default` profile. 
3. That may be true. However, the PTP profile does affect whether the sensor will "jump" time to synchronize if the offset is greater than 1 sec (`default-relaxed`) or simply adjust clock frequency and gradually catch up (`default`).

Please let us know if you have resolved this issue. Thanks.

dgarbuzov, 2021-07-01:
Closing due to inactivity

closed 2021-07-01 by dgarbuzov