ouster-lidar/ouster-sdk #225: Messages get dropped [ros melodic + industrial camera]

afridi26, 2021-02-17:
Hey everyone,

I am trying to perform a ground truth validation, having an Industrial camera and Ouster lidar (os1-64) where  both are connected to Nvidia Agx via PoE switch. The bandwidth rate of the camera is `49 MByte/sec` and Lidar Mode is `1024*20`.

If both the camera and ouster run at the same time, my messages are getting dropped. Whereas, if I only run the ouster client I am getting all the messages. You can see the comparisons below:

I am trying to run with following command to record the messages:
```
rosbag record -b 1024 --chunksize=1024 --duration=1h  /os_cloud_node/points __name:=recordLidarData &
```

This is the output of the rosbag file which was recorded without the camera presence on the network:
You can see the messages are `71991` and size of the recorded bag for 1 hour is almost `210.9 GB`:
```
path:        _2018-01-28-17-35-57.bag
version:     2.0
duration:    59:59s (3599s)
start:       Jan 28 2018 17:35:57.88 (1517157357.88)
end:         Jan 28 2018 18:35:57.60 (1517160957.60)
size:        210.9 GB
messages:    71991
compression: none [71991/71991 chunks]
types:       sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
topics:      /os_cloud_node/points   71991 msgs    : sensor_msgs/PointCloud2

```
When simultaneously, both the camera and ouster clinet runs, I am getting the correct FPS for the camera but all the messages are getting dropped for ouster and ended up in very small bag file. You can see the output of the rosbag file:
Here you can see the messages I am getting for 1 hour are only 6294 and the size is 18.04 GB.
```
path:        _2018-01-28-19-15-51.bag
version:     2.0
duration:    59:58s (3598s)
start:       Jan 28 2018 19:15:52.19 (1517163352.19)
end:         Jan 28 2018 20:15:50.64 (1517166950.64)
size:        18.4 GB
messages:    6294
compression: none [6294/6294 chunks]
types:       sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
topics:      /os_cloud_node/points   6294 msgs    : sensor_msgs/PointCloud2
```

Can anyone nudge me in the right direction how to solve this issue. I mean in terms of, for example. what will be correct chunksize or buffsize so my messages should not drop. Is there anyway, where I can check how many messages are dropped kind of statistical information?

Regards,
Affi

afridi26, 2021-03-09:
The issue was about processing, now solved , thanks.

closed 2021-03-09 by afridi26

dgarbuzov, 2021-12-28:
Think another user might be running into something similar (#298). Could you elaborate on what the issue was?

nuneslu, 2022-02-02:
Hi I'm also having the same problem, could you maybe share how did you solved it?

chinitaberrio, 2022-02-15:
> The issue was about processing, now solved , thanks.

@afridi26  Could you please tell us how you solved this issue? I have the same issue with the same Nvidia computer.