ouster-lidar/ouster-sdk #227: ROS messages going backwards in time

ahaefner, 2021-03-06:
We are having an issue where ROS messages from the ouster point cloud topic are going backwards in time. See the attached plot where both the ros msg header time and the arrival time are plotted from a rosbag. We have seen this issue with both PTP time and the ouster internal clock time.

[image]

We're seeing this issue with an OS1 with firmware 2.0. We did not see this issue previously with firmware 1.13 and the older ros drivers.

dgarbuzov, 2021-04-30:
Is this reproducible? Does it happen at a regular interval? Can you check if the timestamps on the packets are exhibiting the same behavior?

If you send over a packet capture I could also try to take a look.

dgarbuzov, 2021-07-01:
closing due to inactivity

closed 2021-07-01 by dgarbuzov