ouster-lidar/ouster-sdk #228: Cannot get data on ROS from OS0

robokamran, 2021-03-09:
Hello,

I need to use OS0 with ROS. I have compiled `ouster_example` for my Ubuntu 18.04 with ROS Melodic.
I can launch the node with `roslaunch ouster_ros ouster.launch` and I get the topics `/os_node/imu_packets` and so on with `rostopic`, but I get no message published on any of those topics.

I have used this command:
` roslaunch ouster_ros ouster.launch sensor_hostname:=192.168.137.163 udp_dest:=192.168.137.1 lidar_mode:=2048x10 viz:=false`

And here is my self written launch file:

```
<node pkg="ouster_ros" name="os_node" type="os_node" output="screen" required="true">
    <param name="~/lidar_mode" type="string" value="2048x10" />
    <param name="~/timestamp_mode" type="string" value="" />
    <param name="~/replay" value="false" />
    <param name="~/sensor_hostname" value="os-992106000257.local" />
    <param name="~/udp_dest" value="192.168.137.1" />
    <param name="~/lidar_port" value="7502" />
    <param name="~/imu_port" value="7503" />
    <param name="~/metadata" value="" />
</node>
<node pkg="ouster_ros" type="os_cloud_node" name="os_cloud_node" output="screen" required="true">
    <remap from="~/os_config" to="/os_node/os_config" />
    <remap from="~/lidar_packets" to="/os_node/lidar_packets" />
    <remap from="~/imu_packets" to="/os_node/imu_packets" />
    <param name="~/tf_prefix" value="" />
</node>
```

robokamran, 2021-03-17:
The problem was that my ubuntu had a filter on port number "0"
issue solved

closed 2021-03-17 by robokamran

dlee640, 2021-04-13:
Hello. I have the same issue. Can you provide a hint on how you got rid of the filter on port 0? Thank you !!

robokamran, 2021-04-15:
Hi, honestly I had a lot of troubles running the OS0 properly and stably on both Windows WSL and a normal Linux.
Basically the ouster node uses IPV6 to support IPV4 sockets for no reason. It uses TCP for connections, handshakes and configurations, but, uses UDP socket for receiving data. On a theoretical basics it looks like that it should work without any problem, and it's very good that it can support both IPV4 and IPV6. However, on a practical situation, specially when you use hostnames instead of direct IP, it gets confused and connects one socket on IPV4 and the other on IPV6 and that leads to situations like you have to do a system restart after each use or not being able to use the sensor at all! We have multiple ouster scanners on different robots. This is basically what I did:
1- Disable IPV6 on the whole ubuntu.
2- Disable DHCP on ouster and give it an static IP.
3- Set a properly ranged static IP on your Windows or Linux.
4- Replace all IPV6 sockets on ouster node with IPV4 sockets.
5- Remove all IPV6 checks in ouster node (they are useless anyway, not even good for IPV6 sockets).
6- Do the firmware update on the ouster sensor,
7- Avoid setting port number 0 because it is dangerous anyway and use 7502 and 7503.
8- If you're using a proxy system (like WSL2 on Windows), be sure to run a UDP port forwarder (all normal port forwarders are TCP so you may write your own program like I did).

Hope that works for you. I did all that and now I can use the ouster like a normal sensor with multiple shutdowns and startups and different config each time. With the ouster studio and Windows applications it's a complete different story, but for Linux you should do as the procedure I did if you want to have a robust sensor on your robot.

dgarbuzov, 2021-04-30:
Port 0 isn't a real port -- in socket APIs it's used to dynamically select a free port. I'm not sure what "filtering port 0" could mean.

You should be able to turn off ipv6 per interface with either sysctl or NetworkManager. I wouldn't recommend disabling ipv6 in the kernel to work around ipv6 misconfiguration.

robokamran, 2021-04-30:
@dgarbuzov thank you for your message.

Well, by term "filtering port 0" I meant that system has a restriction on allocating dynamically a free port to port number zero.
Honestly IDK the proper term for it :))

And as for IPV6, yes I have disabled the IPV6 from either boot grub or sysctl or NetworkManager, yet when I launch the ouster node, it gives an error regarding it cannot open the socket. I can send the exact error if you need.
When I debugged it, it was throwing an exception on opening the IPV6 socket (which is compatible with IPV4) but since IPV6 config is not enabled by system, it throws an exception. After changing the socket to normal IPV4, it will throw again another exception on the next lines after the socket opening code that are just checking for IPV6 available configs (which is unnecessary even for using IPV6). So that's the case. I can provide exact errors and lines of code which need the edit if you want.

BTW we've tested on different computers and robots running ubuntu 18.04.

dgarbuzov, 2021-04-30:
Yes I've never heard of preventing dynamically selecting sockets as a security policy either. In any case, seems like specifying a port explicitly works.

The client code currently uses "dual stack sockets" which require ipv6 support in the kernel, but not necessarily having ipv6 configured or enabled on the interface. If you remove the `ipv6.disable` boot parameter you may have added in grub, but keep it turned off in either sysctl or NetworkManager, you should be able to use the code as-is without having to deal with ipv6 at all.

robokamran, 2021-05-18:
@dgarbuzov 
Thank you very much for the instructions.
But sometimes if you disable IPV6 through NetworkManager or sysctl, it will be enabled again after a reboot.
You can refer to [this document](https://itsfoss.com/disable-ipv6-ubuntu-linux/) for example. We had the same issue on different computers and robots so mostly we disable it from the grub to make sure it is fail safe. (sorry for the delay of response).