ouster-lidar/ouster-sdk #231: Issue of timestamp of Ouster LiDAR

yulan0215, 2021-03-13:
Hi:
I have a question that I followed the instruction in the weblink: https://forum.ouster.at/d/74-how-to-synchronize-gps-with-ouster-sensor
And I checked the synchronization by TCP. However, when I ran the driver of Ouster LiDAR, the LiDAR was still used the time from the Ubuntu system. Can you give me some ideas how to solve this problem? Thx!

yulan0215, 2021-03-13:
> Hi:
> I have a question that I followed the instruction in the weblink: https://forum.ouster.at/d/74-how-to-synchronize-gps-with-ouster-sensor
> And I checked the synchronization by TCP. However, when I [...]

And I would like to add one more information that my GPS module not only give the message "GPRMC". It also transfer other messages(for instance: GPGGA). I would like to ask is it influence the timestamp of my lidar? The timestamp was still the ROS time. Thanks!

yulan0215, 2021-03-14:
> And I would like to add one more information that my GPS module not only give the message "GPRMC". It also transfer other messages(for instance: GPGGA). I would like to ask is it influence the [...]

I think I solved the problem but one another problem is what command can I input to get the date (in the terminal). Shown below:
[image]

Thx!

Krishtof-Korda, 2021-04-21:
> I think I solved the problem but one another problem is what command can I input to get the date (in the terminal). Shown below:
> [image]
> 
> Thx!

@yulan0215 can you clarify the question? Are you looking for a command to the sensor or a terminal command? Can you explain exactly what you are trying to achieve? Thanks.

dgarbuzov, 2021-07-01:
Haven't gotten a response; please reopen if you're still having issues

closed 2021-07-01 by dgarbuzov

yulan0215, 2021-07-20:
> Haven't gotten a response; please reopen if you're still having issues

I am sorry I still have problem.

dgarbuzov, 2021-07-21:
Are you able to get in touch with Ouster support? That would be a better channel for this question

yulan0215, 2021-07-22:
> Are you able to get in touch with Ouster support? That would be a better channel for this question

Thx! I will contact them later.