ouster-lidar/ouster-sdk #232: OS-16 no data_format found and no lidar_origin_to_beam_origin_found

raghavthakar, 2021-03-15:
Hi! As the title suggests, I am trying to interface an Ouster OS-16 LIDAR. on running the command:
`roslaunch ouster_ros ouster.launch sensor_hostname:=os1-99XXXXXXXXXX udp_dest:=X.XXX.XX.X lidar_mode:=512x10 viz:=true`

The sensor starts up and the visuaizer node starts up just fine. However, the visualizer node shows empty data. I get 2 warnings saying:
```
WARNING: No data_format found.
WARNING: No lidar_origin_to_beam_origin_mm found.
```

Please advise, any help would be great. Thanks in advance!

kwong0, 2021-04-13:
Hi @raghavthakar,

Are you by any chance not running fw 2.0 on your sensor?

x86-64-slc6-gcc47-opt, 2021-05-25:
Hi, I have the same problem with reading .pcap file from ouster with fw 2.0 using Python , can i somehow get around this error?

dgarbuzov, 2021-05-26:
You should also double check that your `udp_dest` is configured correctly: rostopic should should show you if there are packets being received by the ROS nodes, and if not, you can check if there's data arriving on the UDP ports with a tool like wireshark (or tcpdump, or just netcat).

closed 2021-05-26 by dgarbuzov